espruino/BangleApps #1219: Recorder app low mem error makes it unusable

fparri, 2022-01-05:
I keep getting a Low Memory Memory error with the Recorder app running on my Bangle.js and I cannot turn it OFF once I first turned it on. Only solution seems to remove it. Apps Manager says I still  1,746,608 bytes free.

Any ideas? I took screenshots of the error. How can I upload, in case it's needed?

gfwilliams, 2022-01-06:
Argh, thanks! The low memory is referring to RAM. So you don't have issues with Recording off, but when you turn it on you have problems?

What other widgets do you have installed? Is that GPS status, and the Active Pedometer?

fparri, 2022-01-06:
Hi Gordon. I guess it could be a low ram error, as it only happens when I have a certain number of apps on the Bangle JS 1. Thanks a lot :)

gfwilliams, 2022-01-06:
I think the issue will be the widgets you have installed rather than the apps. I seem to remember active pedometer especially is a bit of a memory hog :(

I'll leave this open though as I may be able to find a way of getting Recorder to use less memory

fparri, 2022-01-06:
I see. Thanks a lot again :)

elrod16, 2022-01-07:
This might be a red herring, but I've actually been seeing issues with RAM and a few GPS apps recently on the Bangle2:
-GPS Info very frequently has "FIFO_FULL" or "FIFO_FULL" and "LOW_MEMORY" drawn over each other.
-SensiBLE GPS subsection triggers a "LOW_MEMORY" error that actually crashes the system back to the bootloader given enough time.

gfwilliams, 2022-01-10:
The GPS Info 'FIFO_FULL' error isn't something I'd worry about too much - it's a recent change that got added to GPS Info.

However you're sure you are seeing 'LOW MEMORY' on GPS Info? That's a bit of a worry

elrod16, 2022-01-10:
For GPS Info, only twice ever. And the red console text was superimposed over another error, but both times that happened it crashed back to bootloader

gfwilliams, 2022-02-02:
Just checked on this...

* With Morphing clock and recording on, we get `free: 1746, usage: 838` which I think is more than acceptable for memory usage
* I made a quick tweak and the recorder menu uses 1270 vars which is a bit high, but that's still less than half the RAM on Bangle.js 1.

Closing this as I think that situation is ok

closed 2022-02-02 by gfwilliams