espruino/BangleApps #1235: Assisted GPS working for Bangle.js 2?

myxor, 2022-01-07:
Is the [Assisted GPS app](https://banglejs.com/apps/#assisted) working for the Bangle.js 2?

I made a few tests and it looks like it is working but the app is marked as Bangle.js 1 only and the descriptions says the same - so i am not 100% sure.

If it is confirmed working on B2 i can update the app info and description.

gfwilliams, 2022-01-07:
No, I'm afraid not - while it uploads data that data is not understood/ignored by Bangle.js 2's GPS.

There's a thread about progress at http://forum.espruino.com/conversations/371360/#comment16336280

gfwilliams, 2022-01-12:
I posted this up in some previous conversation, but had trouble finding it now, so: https://github.com/geary/AnyTone-D868UV/issues/61

Some interesting info about improving time to GPS lock there

gfwilliams, 2022-01-12:
Original post: http://forum.espruino.com/conversations/369403/#comment16292936

elrod16, 2022-01-13:
@gfwilliams I just tried the A-GPS update you put up and wow, it's a ridiculously huge difference.

elrod16, 2022-01-13:
Yeah, it normally takes 5-10 min per fix with 4-6 satellites (standing outside). Where I live is very cloudy and overcast all the time so I just wrote the performance off as understandable considering what a small and low power device it is and the poor visibility.

After trying your updated AGPS setup though, I got a fix indoors, by a window, in < 30 seconds with 9 satellites. I tried it from a cold boot too and it still was working great. Tested with GPS Info, My Location, Speedo, and GPS Time.

pelrun, 2022-01-15:
FYI the AGPS update data we get only has GPS ephemerides in it. The selection is independent and actually enables/disables use of the constellations entirely. So it's probably a good idea not to select one without GPS in it :) There's some anecdotal evidence that disabling BDS helps if you're not in the Asia-Pacific region, but it's hard to verify that.

elrod16, 2022-01-15:
So which constellations does the device use by default (if I understood you correctly)? Just curious because I set up another factory fresh B2 with no AGPS but updated to the latest firmware and compared it head to head with my B2 that has the AGPS update. The updated AGPS B2 consistently had way higher numbers for "sats in view" and the end number of sats it got its fix from (though not as extreme as the difference between "sats in view").

gfwilliams, 2022-01-17:
By default it's using GPS and BDS I believe (but I could be wrong). I *think* selecting just GPS allows it to use the other repurpose the receivers that it would have been using for BDS onto GPS.

It's possible that if you're still using GPS/BDS/GLONASS then the satellite count actually includes GPS+BDS+Glonass satellites? Although 9 is still totally plausible just for GPS.

Feel free to discuss here, but I'm closing this issue now that AGPS seems to work fine :)

closed 2022-01-17 by gfwilliams

elrod16, 2022-01-17:
Thanks for the info, when sending the file to the B2, where does it have to be targetted for it to recognize it and apply it to the GPS receiver? 

I ended up doing a bunch of side by side comparisons between the receiver at factory default and the one I'd been tweaking, turns out BDS wasn't doing anything where I am, GLONASS was where I got the performance boost. GPS alone barely had enough satellites in view to even attempt a fix.

gfwilliams, 2022-01-17:
Very interesting. I hadn't even considered GLONASS would be faster!

> Thanks for the info, when sending the file to the B2, where does it have to be targetted for it to recognize it and apply it to the GPS receiver?

Not sure I understand. If you look at the AGPS `custom.html` app the filename is set to `RAM` which just tells the app loader to send the commands as-is. Then the commands are `Serial1.write...` which writes to the GPS

elrod16, 2022-01-17:
Thanks, that's exactly what I wanted to know