espruino/BangleApps #1249: infinite buzz in response to Bangle.setUI("clockupdown", btn=> { draw(); }); after a settings reload

hughbarney, 2022-01-08:
When using a clock that uses Bangle.setUI("clockupdown",) you sometimes get an infinite buzz when pressing top or bottom.
I have tracked this down and can reproduce every time by just by running the settings app and causing a ''reloading ..." message to be displayed.  Then in the default clock that uses Bangle.setUI("clockupdown",)  tap the top right or top left, you will start an infinite buzz() and the only way to stop it is to do a BTN1 long press reset.

I started to notice it again after upgrading to v2.11.07.  It may not be specific to v2.11.07 as I have experienced this on 2v10.x firmwares at times as well.  But never been able to pin it down.

Here's the exact config and steps to reliably reproduce:
1) perform a factory reset from the settings menu
2) When it reboots, step through the welcome screen
3) Install Simplest Clock (any version) - this will be overwritten later with the test code.
4) Set simplest clock as the default clock through settings
5) Connect to the App Loader and delete all apps except 
      BootLoader 0.39  Launcher v0.10 Setting v0.37  (these are the app versions on factory reset)
 This step establishes that it is no other app that is causing the problem (eg I was experimenting with wdigets and had to exclude them from the config).
6) Now upload the simplest.app.js file attached (this is just a modified version of simplest but with these additional lines of code.

```
Bangle.setUI("clockupdown", btn=> {
  draw();
});
```

7) Restart and establish that simplest clock is working and that when you tap on top or bottom right it only produces the short feedback buzz that you would expect.

8) Now go into Settings through the launcher and change a setting that causes a 'reloading ...'  message.
For example change the theme from light to dark, or switch BLE off.  Exit the settings app.

9) The simplest clock should be displayed as it is the default clock.
Now tap on the top or bottom right and you should experience an infite buzz.

TEST CODE below GIT HUB wont let me attach

```
const h = g.getHeight();
const w = g.getWidth();

function draw() {
  var date = new Date();
  var timeStr = require("locale").time(date,1);
  
  g.reset();
  g.setColor(g.theme.bg);
  g.fillRect(Bangle.appRect);

  g.setFont('Vector', w/3);
  g.setFontAlign(0, 0);
  g.setColor(g.theme.fg);
  g.drawString(timeStr, w/2, h/2);

  queueDraw();
}

// timeout used to update every minute
var drawTimeout;

// schedule a draw for the next minute
function queueDraw() {
  if (drawTimeout) clearTimeout(drawTimeout);
  drawTimeout = setTimeout(function() {
    drawTimeout = undefined;
    draw();
  }, 60000 - (Date.now() % 60000));
}

// Stop updates when LCD is off, restart when on
Bangle.on('lcdPower',on=>{
  if (on) {
    draw(); // draw immediately, queue redraw
  } else { // stop draw timer
    if (drawTimeout) clearTimeout(drawTimeout);
    drawTimeout = undefined;
  }
});

g.clear();

// Show launcher when middle button pressed
//Bangle.setUI("clock");
// use  clockupdown as it tests for #
Bangle.setUI("clockupdown", btn=> {
  draw();
});

// Load widgets
Bangle.loadWidgets();
Bangle.drawWidgets();
draw();
```

hughbarney, 2022-01-08:
Note can also reproduce with 2v11.07 booloader 0.40 and settings 0.40

HilmarSt, 2022-01-09:
I don't know if it's the same mechanism, but I've got an infinite buzz (actually a Bangle2 beep mapped to buzz) while using the "Hour Chime" App for a while and changing the App's setting from buzz to beep.

gfwilliams, 2022-01-10:
Thanks! Just reproduced. This is really helpful!

gfwilliams, 2022-01-10:
Ok, there's some strangeness caused by the buzz being issued while the Bangle is reloading apps...

So for instance `Bangle.buzz();load()` means that the next `Bangle.buzz()` gets stuck on or doesn't do anything.

I've now fixed this, and *hopefully* that means it's not going to happen in the other cases either, but I guess we'll see!

closed 2022-01-10 by gfwilliams

hughbarney, 2022-01-10:
May not be fixed see:
http://forum.espruino.com/conversations/371717/

gfwilliams, 2022-01-11:
Yes, I saw that - interested to see if anyone else has issues since so far it's working great for me