espruino/BangleApps #1251: GPS Info FIFO_FULL error

elrod16, 2022-01-09:
FIFO_FULL error until the GPS gets a fix. Also mentioned in #1219. On Bangle 2 fw version 2v11

HilmarSt, 2022-01-09:
FIFO_FULL is caused by Bangle.on('GPS-raw', onGPSraw) :
GPS-raw is spilling out (too) many characters per second, I currently don't know how to avoid FIFO_FULL.
Hopefully someone can tell me how to temporarily increase the FIFO buffer.

gfwilliams, 2022-01-10:
Ok, so this is the GPS Info app? It's only recent versions where GPS-raw is used.

Luckily the FIFO_FULL message doesn't come because the FIFO is actually full in this case, but it's because the GPS info line wasn't handled before the next GPS info line came in - so basically it just means one GPS info line got lost.

... but unfortunately that means the buffer size can't be increased. The best way to fix this would be to try and make sure there is no code that takes too long to execute. I guess it could be the rendering code that's the issue in this case?

elrod16, 2022-01-10:
Interesting. The UI seems pretty basic though. Could background processing of apps/widgets play a role? Or is the foreground app prioritized enough to avoid that?

BartS23, 2022-01-10:
I was able to reduce the FIFO_FULL messages to one at the app start.

1. the lazy layout rendering takes too much time.
2. enable the gps at the end of the file.
3. save SATinView to detect changes.

https://github.com/BartS23/BangleApps/commit/6eb4c8295866aa7540b66cd08c13df62c666c262?diff=split

Can anyone confirm this?

gfwilliams, 2022-01-10:
Those changes look pretty promising

HilmarSt, 2022-01-10:
On my Bangle2:
- the FIFO_FULL messages in the IDE are reduced to 2 at start-up (instead of 1 per second)
- the output of `layout.progress.label = "in view: " + SATinView;` is **missing** on the screen
- FIFO_FULL is still written on the screen

BartS23, 2022-01-10:
I do not know why, but only half of my change was committed :(

@HilmarSt could you please try again with the new commit https://github.com/BartS23/BangleApps/blob/7e14d51188f90981145813e38a7f50f5fde685b7/apps/gpsinfo/gps-info.js

HilmarSt, 2022-01-11:
looks good now:
- only 1 FIFO_FULL message in the IDE
- no FIFO_FULL any more on the screen

I would say: Thank you, ready for PR.

gfwilliams, 2022-01-12:
I think the recent change should help with this - and on the most recent firmwares I have now removed lost GPS data creating a FIFO_FULL error, and it now reports lost packets as a second argument to the GPS-raw event itself

closed 2022-01-12 by gfwilliams

HilmarSt, 2022-01-12:
@gfwilliams Any chance to mention it on https://www.espruino.com/Reference#l_Bangle_GPS-raw ?

gfwilliams, 2022-01-12:
Yes, it will automatically update with the next firmware release