espruino/BangleApps #1252: SensiBLE memory error and system crash

elrod16, 2022-01-09:
SensiBLE will rapidly consume RAM and cause LOW_MEMORY when on the GPS sub menu. Given enough time it will crash the system to a full restart from boot loader. Bangle 2 fw version 2v11. Also mentioned in #1219.

gfwilliams, 2022-01-10:
@jeffyactive  ?

jeffyactive, 2022-01-11:
Thanks for raising the issue @elrod16, and thanks for flagging it to me @gfwilliams.

v0.05 of SensiBLE implements all the target functionality, but hasn't yet been tweaked for performance and long-term stability.  As it runs all the sensors on the Bangle, and can update their values in near-real time on the display, and transmits plenty of BLE advertising packets, it is quite demanding on the computing resources.  I've observed the crashes during development too.

I don't have bandwidth to dive into this short term, but if you have a specific recommendation of what to fix in the code, I'll try to squeeze it in.

**Suggested temporary workaround if you just need GPS**: go into the other sub menus and disable the other sensors, which will free up resources.

gfwilliams, 2022-01-11:
Thanks - there's definitely something odd going on as I just tried and it seems reliable for me and no extra memory is being used.

But... When the BLE connection is active it seems to crash after a while with a watchdog error (lockup). I don't think it's anything you are doing @jeffyactive - it must be something inside the Bangle.js firmware.

@elrod16 I think you reported a LOW_MEMORY error with another GPS app where nobody else sees it, and I'm wondering now - do you have some other apps/widgets installed that might be causing this problem instead?

elrod16, 2022-01-11:
@gfwilliams Yeah I'm suspecting the same thing. Over in the GPS Info issue thread i wrote:
> Interesting. The UI seems pretty basic though. Could background processing of apps/widgets play a role? Or is the foreground app prioritized enough to avoid that?

It seems like something I updated recently isn't playing nice. When I get a moment I'm going to backup my cfg files on the watch, erase it all, and incrementally add each fw/system app update and then move on to installing every 3rd party app (ones that are showing mem issues first) and run a test scenario that has good consistency for starting the crashes.

Let me know if there are any specific test scenarios you want or any specific data to be pulled via IDE.

HilmarSt, 2022-01-25:
With SensiBLE v0.05 and FW 2v11.34 there are some error messages like
```
Interpreter error: [
  "MEMORY_BUSY"
 ]
New interpreter error: MEMORY_BUSY
```
and
```
Uncaught [object Object]
 at line 1 col 389
...owName:false,interval:interval});
                                  ^
in function called from system
```
on the console and sporadic watchdog resets.

It seems that **all** sensors are active by default, e.g. the HRM is active when still in the app menu.

Maybe it's better to disable all sensors by default - I can offer to implement and test this.

jeffyactive, 2022-01-25:
Thanks for the feedback @HilmarSt, just the fact that SensiBLE seems to be getting used long enough to have it crash is motivation enough to make some improvements myself!

I had originally planned to use the [Data Storage](https://www.espruino.com/Bangle.js+Storage) feature to remember the user's choices of which sensors to activate.  I think I will also set the sensors to be inactive by default, as you suggest, save for the accelerometer which is on anyway.

I'd nonetheless be interested to know what exactly in the code is overwhelming the Bangle, whether it is the sheer number of events or something I might improve in my code.  Please don't hesitate to share your findings, thanks!

jeffyactive, 2022-01-25:
Just made a pull request for v0.06 which does indeed:
- disable all sensors (except the accelerometer) on initial load
- save the user settings of which sensors to enable/disable to persistent storage, so that they are recalled whenever the app is run

@gfwilliams, is it possible that the combination of running the Bluetooth UART service in parallel with the SensiBLE advertising data results in instability?  During development I noticed several crashes when SensiBLE was running while connected to the IDE or App Loader, but few if any when SensiBLE ran standalone.  :thinking:

gfwilliams, 2022-01-25:
> is it possible that the combination of running the Bluetooth UART service in parallel with the SensiBLE advertising data results in instability?

It shouldn't... I mean, when you're connected the advertising should just get ignored until you disconnect.

Do you still get the crashes when not all the sensors are running?

HilmarSt, 2022-01-25:
> Do you still get the crashes when not all the sensors are running?

- I've installed v0.06 - only Acceleration is "On" by default.
- go to Barometer and switch it on:
```
Interpreter error: [
  "LOW_MEMORY",
  "MEMORY_BUSY"
 ]
```
- stay in the Barometer for some seconds:
Watchdog reset.

When I look into the code I coulnd't find a delay for the sensor readings, does that mean they are updated "as fast as possible"?
I would propose a minimal update rate of about 200 milliseconds.

Why is it not possible to switch off Acceleration? That would help finding the app's issues.

gfwilliams, 2022-02-03:
Do you get that `"LOW_MEMORY",` error almost immediately? Because I'm trying it here and it's fine for me.

> When I look into the code I couldn't find a delay for the sensor readings, does that mean they are updated "as fast as possible"?

I believe they are updated whenever there is a new reading from the barometer, which seems fair enough.

I just narrowed the MEMORY_BUSY error down to:

```
const APP_ADVERTISING_DATA = [ 0x12, 0xff, 0x90, 0x05, 0x7b, 0x6e, 0x61, 0x6d,
                               0x65, 0x3a, 0x73, 0x65, 0x6e, 0x73, 0x69, 0x62,
                               0x6c, 0x65, 0x7d ];
setInterval(function() {
  let data = [ APP_ADVERTISING_DATA ]; // Always advertise at least app name
  data.push([ APP_ADVERTISING_DATA ] );
  NRF.setAdvertising(data, { showName: false });
  process.memory() // <---- errors here
}, 1000);
```

So something in setAdvertising is messing with the memory. It's nothing to do with the barometer other than that it causes an extra field to be added to what's set in setAdvertising. 

If you do `const APP_ADVERTISING_DATA = new Uint8Array([ ...` then it actually fixes the problem. So, my guess is it's how Espruino handles 'rotated' advertising

gfwilliams, 2022-02-03:
Fixed with https://github.com/espruino/Espruino/commit/7c2bb3265ce8fa80a54864eb7a273f05e2360f4e

closed 2022-02-03 by gfwilliams