espruino/BangleApps #1255: [LCARS clock] reduced functionality with full screen mode

CountMurphy, 2022-01-09:
First off I'd like to say that I LOVE this clock face. I also love the new update that introduces multiple screens. Thank you for the work that was put in. 

Since the clock is now presented in full screen mode, none of the installed widgets are visible. Most of these widgets don't really matter so much since LCARS will display the same content already (battery level, step counter, GPS status etc). 

However, both weather data, and missed message notifications from the messages app are not shown while in full screen mode. Second to the clock itself, I find I'm checking the watch for weather most often, and having it missing from the main display is an inconvenience. I have two ideas that could address this:

1. make full screen mode user configurable
2. add weather data as an option for the rows (following the systems Celsius vs Fahrenheit preferences)
    and add icon for missed messages

I think option 1 would be easiest/quickest. Thanks again for this amazing interface!

adamschmalhofer, 2022-01-10:
Option 1 isn't as easy/quick as it might first appear. The Layout is hand crafted including the size of the fonts. The easiest way would be to drop one of the data-rows and the "STATUS"/"ALARM"-text.
Option 2 is certainly in the spirit of LCARS

CountMurphy, 2022-01-10:
Thanks for the reply. Redrawing the entire interface is definitely not worth doing for something as minor as this. If option 2 is something worth doing, I can start reading up on bangle development to help contribute.

gfwilliams, 2022-01-10:
Just to add there is another option... You can relocate the position of the widgets pretty easily, so potentially you could hide all the widgets that you know you handle already (pedometer, etc) and then *if* there are others you could move them to some part of the screen that's not quite as useful as the others (like the vref bit?)

Could be an option anyway - rather than playing cat and mouse trying to add all the features from every widget to the clock