espruino/BangleApps #1257: BT HRM

f-teacher, 2022-01-10:
I'm using a BT HRM (polar 0H1+) and I'm glad @halemmerich updated the BT HRM app on the banglejs2. I tried it straight away and for my testing, I wore my Polar HR working on my arm.
When the settings of the app are BT HRM ON and HRM event OFF, if I start the BT HRM app, I get 2 measures (event HRM and event BTHRM).
When the settings of the app are BT HRM ON and HRM event ON, if I start the BT HRM app, I get only 1 measure (event HRM, which are figures from the wrist), and the HR widget gives measurements 3 times per seconds).
I suspect something is wrong somewhere.

halemmerich, 2022-01-10:
I think you see mixed readings from both sensors happen. If other apps using the HRM were installed prior to the BTHRM app and those apps activate the HRM in the boot0 code, then currently there is nothing that prevents the internal HRM reporting the events to all apps. I have done a little experiment at [hal](https://halemmerich.github.io/BangleApps/#bth). This code deactivates the internal HRM for all apps and moves it over to BTHRM.
Let me know if this helps with your issue, then I can clean it up and create a pull request.

gfwilliams, 2022-01-11:
I had thought that at least originally BTHRM ran at boot time, which would mean it could jump in before the HRM was properly initialised and ensure that it didn't get turned on?

It's easy enough to check if that's the case though - does the onboard HRM LED flash green when you finger is near?

halemmerich, 2022-01-11:
BTHRM still runs at boot time, but at least the health app does too and calls setHRMPower() for itself. In case BTHRM runs second, the first setHRMPower() is not modified and the internal HRM is turned on. Which then results in BTHRM and HRM events being mixed.
[This commit](https://github.com/halemmerich/BangleApps/commit/c3482fa1107c01d96568ce6566f186388e1057e3) tries to remedy that. I do not like how it changes the effects of other apps HRM calls after the fact, but since the calls itself are redirected that is probably valid. It should be enough to do this once during boot after modifying the HRM calls instead doing it on every call as the commit currently does.

gfwilliams, 2022-01-11:
Hmm. Just looking at https://github.com/halemmerich/BangleApps/commit/c3482fa1107c01d96568ce6566f186388e1057e3 - it seems all the checks are for `settings.enabled`?

So what if  `settings.enabled=false`? Shouldn't we just be using the internal heart rate monitor? It seems in this case even the internal HRM is disabled?

If the issue is the `health` app, maybe we should modify that such that the `boot.js` turns the HRM on after a slight delay using `setTimeout`? Then we can avoid this issue without too much fiddling around

halemmerich, 2022-01-11:
Yes, you're right, this shouldn't work...
I tried adding some additional prints and can not explain why it actually still does work:
```
--- HRM Power ---
Settings: {"enabled":false,"replace":false}
isOn: 0, app hralarm
Disable BT HRM
Disable INT HRM
--- HRM Power end ---
--- HRM Power ---
Settings: {"enabled":false,"replace":false}
isOn: 1, app circleclock
--- HRM Power end ---
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v11 (c) 2021 G.Williams
--- HRM Power ---
Settings: {"enabled":false,"replace":false}
isOn: 0, app hralarm
Disable BT HRM
Disable INT HRM
--- HRM Power end ---
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v11 (c) 2021 G.Williams
--- HRM Power ---
Settings: {"enabled":false,"replace":false}
isOn: 1, app bthrm
--- HRM Power end ---
--- HRM Power ---
Settings: {"enabled":false,"replace":false}
isOn: 0, app hralarm
Disable BT HRM
Disable INT HRM
--- HRM Power end ---
Event for Int {"bpm":96,"confidence":0,"history":[0,0,0,0,0,0,0,0,0,0,24.5,93.7,25.2,65.9,100,23.5]}
Event for Int {"bpm":88,"confidence":0,"history":[0,0,0,0,0,0,0,0,0,24.5,93.7,25.2,65.9,100,23.5,73.1]}
```
I see only disable calls, but there are actually HRM events coming in, blinking sensor included.

halemmerich, 2022-01-11:
Reboot "fixes" this, it does not work as expected ;) Something I did to my bangle must have caused confusion. [Commit](https://github.com/halemmerich/BangleApps/commit/4899115f3f8565bbc0a7a603109d4c0d9ff4602c) seems to work better.

halemmerich, 2022-01-13:
Should be fixed with https://github.com/espruino/BangleApps/pull/1286

f-teacher, 2022-01-16:
I'm not sure I got all this right. I'm using  firmware 2v11.24 and the BTHRM v0.03. In the settings section of the BT HRM app, the BT HRM works as expected.
The HRM event (which seems to relate to the wrist HRM) does the opposite of what it says.
Precisely, with the watch on my wrist and the BT HRM on and on my arm. I switch HRM event **OFF**. I go to the app and I can see both the wrist HRM and the arm one. I go back to settings, I swith HRM event **ON** and I can see only my BT HRM.
Is that the expected behaviour ? Because from a end user perspective, it looks wrong.

halemmerich, 2022-01-19:
I have checked the behavior and find it to be as expected (at least it meets my expectations 😁)

```
Setting           Device   App
Enabled   Event   BTHRM    HRM_int HRM_bthrm BTHRM
---------------------------------------------------
off       off     off      yes     no        no
off       off     on       yes     no        no
on        off     off      yes     no        no
on        off     on       yes     no        no
on        on      off      no      no        no
on        on      on       no      yes       no
```

The HRM event settings means that the internal sensors event is to be replaced by the BT sensor event. A point could be made for only replacing the internal sensor if the BT sensor is actually active. Or maybe always creating the BT sensor event and additionally replacing the internal if and when the BT sensor is available.
Since my use case is currently only recording both internal and BT sensor through the recorder app it would be helpful to get other relevant use cases.

gfwilliams, 2022-02-03:
Thanks! I guess this is fixed then :)

closed 2022-02-03 by gfwilliams