espruino/BangleApps #1265: locale-app: discussion: german month abbrevations

DDDanny, 2022-01-11:
Hi,
I'm quite unsure, from programming perspective I like even balanced abbrevations.
https://github.com/espruino/BangleApps/blob/6adb217512c5924612c3341387cc290dd9d3df0b/apps/locale/locales.js#L153

From the german "duden" there are official abbrevations for some month - the rule seems to be that some abbrevations have 3-5 letters (including the traling dot)

Abbrevation including link to "duden":
[Jan.](https://www.duden.de/rechtschreibung/Jan_)
[Febr.](https://www.duden.de/rechtschreibung/Febr_) <<5
[März](https://www.duden.de/rechtschreibung/März)
[Apr.](https://www.duden.de/rechtschreibung/Apr_)
[Mai](https://www.duden.de/rechtschreibung/Mai) <<3
[Juni](https://www.duden.de/rechtschreibung/Juni)
[Juli](https://www.duden.de/rechtschreibung/Juli_Monat)
[Aug.](https://www.duden.de/rechtschreibung/Aug_)
[Sept.](https://www.duden.de/rechtschreibung/Sept_)
[Okt.](https://www.duden.de/rechtschreibung/Okt_)
[Nov.](https://www.duden.de/rechtschreibung/Nov_)
[Dez.](https://www.duden.de/rechtschreibung/Dez_)

What do you think?
* Keep the unoffical 3 letters
* Switch to 3-4 letters without trailing dot
* Switch to 4-5 letters with trailing dot

Cheers,
Danny

HilmarSt, 2022-01-11:
I would like to keep the "unoffical 3 letters".
In my opinion
- always 3 letters are good and fast readable and I think are most commonly used
- it will fit in all layouts even when the developer hadn't thought about localisation

What do you think about exchanging "Mär" with "Mrz"?
"Mär" is an old german word that means something like story or fairytale (Märchen) and I think of it every time I read it.

BartS23, 2022-01-11:
Yes, you are right. 
But there is also DIN 5008 (https://www.din-5008-richtlinien.de/startseite/datum/).

This states that months are abbreviated as follows:
Jan., Feb., Mär./März, Apr., Mai, Jun./Juni, Jul./Juli, Aug., Sept., Okt., Nov., Dez.

So I think the abbreviations used so far are OK.

Regards 
Romek

DDDanny, 2022-01-11:
I already visited another part of the din 5008 iso page.
Thanks @BartS23 for the link
So without the dot it is fine for me to keep this kind of official abr.

closed 2022-01-12 by DDDanny