espruino/BangleApps #1266: Quiet mode: schedule theme?

myxor, 2022-01-11:
What do you think about scheduling the theme via quiet mode? 
I am thinking about switching to dark theme during the night times.

Pinging @rigrig  because he wrote the quiet mode schedule app/widget.

rigrig, 2022-01-11:
I think it would be pretty neat, but to actually make it work (properly) will require some work:
- Switching the theme while running any app isn't as simple as changing a few options
- Ideally I think we should improve the theme chooser first, so we have a "proper" list of themes to choose from, and `qmsched` could let you pick from that (instead of just hardcoding a copy of the default dark/light themes). 

I'm thinking along the lines of extending the `setting` app with
1. `setting.themes` which would hold a list of themes (future idea: load+combine all `*.themes` files)
2. a `setting` library file, so we could do something like `require('settings').setTheme(<theme name>)`, and that would do it's best to switch the theme.

But for switching the theme I think we might really need some general way to tell all apps to redraw themselves...

gfwilliams, 2022-01-12:
Yes... As @rigrig mentioned it can be tricky changing theme because there's no way to trigger a redraw for an app.

However, it *could* do something like change the theme (so next time you change app it takes effect) and if you're in the clock (which 99% of the time you will be) it could just reload the whole clock.

> a setting library file

There was something else recently (hiding widgets I think?) where I thought it'd be worth having a library too. My gut feeling is if we have one library, it'll quickly grow with all kinds of stuff and may end up being a bit of a memory hog esp on Bangle.js 1.

Maybe we could start having smaller libraries in the `modules` folder - like one specifically to set theme, one for widgets, etc.

BUT I think in this case, changing theme is probably easy enough it doesn't require a library. While there's a bunch of code for it in Settings, that's basically only there as a hack to change the theme while in `E.showMenu`, and it wouldn't work reliably outside of that environment. If all you're doing is changing settings.theme it'll be pretty straightforward (although I guess default theme colours would be duplicated?).

rigrig, 2022-01-12:
> However, it _could_ do something like change the theme (so next time you change app it takes effect) and if you're in the clock (which 99% of the time you will be) it could just reload the whole clock.

That seems like a neat solution.

> changing theme is probably easy enough it doesn't require a library

Yeah, it is if we just change the setting and let `load()` handle it.

I was overthinking things, for a start this seems good enough:
- a setting to enable theme switching
- switches between the default dark/light themes by changing `setting.json` (i.e. incompatible with custom themes)
- calls `load()` if running a clock, otherwise just wait for user to switch apps

> My gut feeling is if we have one library, it'll quickly grow with all kinds of stuff and may end up being a bit of a memory hog esp on Bangle.js 1.
>
> Maybe we could start having smaller libraries in the modules folder - like one specifically to set theme, one for widgets, etc.

I meant adding it to the `setting` app, not as a module, so it could run directly from flash. Or does that still hog memory?    
(Now that I think of it, obviously. I was thinking "stored on flash"="free", but it stills need to keep a list of defined functions in RAM.)     
Anyway, I agree that for just this we wouldn't need a library.

gfwilliams, 2022-01-12:
Ok, sounds great!

> I was thinking "stored on flash"="free", but it stills need to keep a list of defined functions in RAM

Yes, sadly that's the problem... Each defined function in a library uses a few more vars

commit 73062edf (2022-01-13): qmsched: switch theme

fixes #1266