espruino/BangleApps #1267: messages app stucks in "no messages with OK button"

DDDanny, 2022-01-11:
This behaviour was described with another topic in #1186

How to reproduce?
1. Connected to ios / android
2. Receive a message
3. Manually Disconnect 

_Root Cause_:
Typically the messages app closes with 
```javascript
checkMessages({clockIfNoMsg:1,clockIfAllRead:1,showMsgIfUnread:1})
````

But on Disconnect the messages are physically deleted
```javascript 
NRF.on("disconnect", () => require("messages").clearAll()); // remove all messages on disconnect
```
to onMessagesModified()
https://github.com/espruino/BangleApps/blob/22420ea269732f783282712cc3aa977fd9b3cef3/apps/messages/lib.js#L60-L74
to showMessage(msg&&msg.id); 
https://github.com/espruino/BangleApps/blob/1d793a037faaca53b86046ecf3e0149b4b610730/apps/messages/app.js#L53-L60
to
https://github.com/espruino/BangleApps/blob/1d793a037faaca53b86046ecf3e0149b4b610730/apps/messages/app.js#L201-L203

And this line 203 I would expect to go home on no message - like the comment:
```javascript
if (!msg) return checkMessages({clockIfNoMsg:1,clockIfAllRead:0,showMsgIfUnread:0}); // go home if no message found
```
But I'm not sure if this will cause side effects.

@gfwilliams: Sorry to mentioned but can you check the above flow and check if L203 can be changed to exit the messages app?
Btw. is it correct/preferred to create this as a issue or should I in future directly comment the code/commit?

gfwilliams, 2022-01-12:
Thanks - I don't see a huge problem with changing that - will push it now.

closed 2022-01-12 by gfwilliams

DDDanny, 2022-01-12:
tested, works like expected: Thanks! 👍

commit 5e353b67 (2022-01-12): messages : If showMessage called with no message (eg all messages deleted) now return to the clock (fix #1267)