espruino/BangleApps #1272: Touch wake-up sensitivity on Bangle.js 2 and possible self-lockout

HilmarSt, 2022-01-12:
When I'm unlocking the LCD with "Wake on Touch" mostly multiple touches are needed.
Can someone please tell me the location of the code so that I can understand the unlock mechanism.

Accidentially I've disabled all 4 unlock sources in the settings menu and ended up with an inoperable watch. What should a "normal" customer do who has no possibility to do a Bangle.setLocked(0) in the IDE?
I think somehow one should be prevented from disabling all screen unlock sources.

gfwilliams, 2022-01-12:
> When I'm unlocking the LCD with "Wake on Touch" mostly multiple touches are needed.

It's done via the the accelerometer since the touchscreen is off - the code is in jswrap_bangle.c in the Espruino repository.

So for a tap, what you need is to actually 'tap' the screen to shake the watch slightly - not just push your finger against it

> Accidentially I've disabled all 4 unlock sources in the settings menu and ended up with an inoperable watch. 

Huh, didn't even think of that :) You just connect with BangleApps and install default apps worst case I guess. Definitely seems like a good idea to stop someone disabling all the options (maybe always re-enable BTN)

HilmarSt, 2022-01-12:
> (maybe always re-enable BTN)
or remove the menu entry to disable BTN?

HilmarSt, 2022-01-12:
Maybe rename "Wake on Touch" to "Wake on Knock" or "Wake on tip" - that seems to work more reliable.

commit 5ea2cf8b (2022-01-12): settings 0.41: Stop users disabling all wake-up methods and locking themselves out (fix #1272)
closed 2022-01-12 by gfwilliams