espruino/BangleApps #1273: [Feature request] Possibility to sort by "New" and "Updated" in non-default App Loaders

HilmarSt, 2022-01-12:
I'm currently using 3 App Loaders:
- https://espruino.github.io/BangleApps/
- https://hilmarst.github.io/BangleApps/
- https://banglejs.com/apps/

Only the last one has the possibility to sort by "New" and "Updated" - if I want to see what's new, I have to look through the closed pull request.
It would be nice to have these sort possibilities also on the non-default App Loaders.

gfwilliams, 2022-01-12:
Answered in your duplicate post here: http://forum.espruino.com/conversations/371794/#comment16347776

HilmarSt, 2022-01-12:
Funny that "TheDod" opened it in the forum at the same time - but it's not me!

closed 2022-01-12 by HilmarSt

gfwilliams, 2022-01-12:
Heh - sorry! It was *so* close I just assumed it's the same :)