espruino/BangleApps #1276: OpenStreetMap: tiles without hillshade?

gfwilliams, 2022-01-12:
Currently we use opentopomap tiles for Openstreetmap:

https://opentopomap.org/#map=17/53.06828/-4.07828

For comparison, Snowdon again but with Openstreetmap: https://www.openstreetmap.org/node/8755297244#map=18/53.06841/-4.07670

It's neat because the colours render better in 3 bit and it also has contours which is good for walking

BUT when you have a decent size hill the hillshade darkens the map such that when converted to 3 bit you just get a big black area.

* Ideally we'd just render in the browser using colors we defined that look good on Bangle.js but that doesn't really seem possible at the moment.
* Maybe somehow we can get tiles without hillshade but with contours, but this seems tricky too.
* Or maybe we can do some filtering on the image to remove the hillshade (after all it is just grey)

gfwilliams, 2022-01-12:
Example:

[image]

myxor, 2022-01-13:
A good page to try out map styles available for leaflet is this one: https://leaflet-extras.github.io/leaflet-providers/preview/

But please note that a lot of these styles require a registration and API Key and some of them have an usage policy which we have to obey.

Just wanted to leave this here so it can help us discussing this.

gfwilliams, 2022-01-13:
Thanks - this is very handy! Makes me think there should maybe be some map style options too...

It seems very hard to find contours without hillshade, so some kind of preprocessing filter might still make sense I guess.

myxor, 2022-01-19:
I played around with the different styles on the preview page and found these three promising:

I made screenshots and converted them using TheGIMP to 8 colors (3-bit):

# Esri WorldTopoMap
[image]

Pricing: free
Info about it: https://www.arcgis.com/home/item.html?id=7dc6cea0b1764a1f9af2e679f642f0f5

# CyclOSM
[image]

Pricing: free
Info about it: https://www.cyclosm.org

# Thunderforest OpenCycleMam
[image]

Pricing: free until 150k tiles / month
Info about it: https://www.thunderforest.com/pricing/

gfwilliams, 2022-01-19:
> I made screenshots and converted them using TheGIMP to 8 colors (3-bit):

Did you do this with a custom palette? Because Bangle.js 2 can only output red/green/blue on/off you really need to threshold each channel, not just choose the 8 most-used colours...

myxor, 2022-01-19:
You are right.
I am no expert in this but i gave it another try with a custom palette of 8 colors (red, green, blue, magenta, cyan, yellow, white and black).
The palette looks like this in css:

```
/* Generated with GIMP Palette Export */
 { color: rgb(255, 0, 0) }
 { color: rgb(0, 255, 0) }
 { color: rgb(0, 0, 255) }
 { color: rgb(0, 255, 255) }
 { color: rgb(255, 0, 255) }
 { color: rgb(255, 255, 0) }
 { color: rgb(255, 255, 255) }
 { color: rgb(0, 0, 0) }
```

I hope this is correct.

The maps now looks like these:

# Esri WorldTopoMap
[image]

# CyclOSM
[image]

# Thunderforest OpenCycleMap
[image]

gfwilliams, 2022-01-19:
Ahh - yes, that's what I'd expect. So what I do with the other tileset is I darken it slightly first.

But that brings back the main issue that the hillshade (when everything is darkened then the hillshade ends up black). But that has just made me think:

Maybe we could convert the RGB to HSB, then increase the saturation to 100%. We basically only care about the color, not the brghtness - so that could do it for us

gfwilliams, 2022-01-19:
If you try now hopefully you will see a definite improvement. Small text is still not great - hard to know how to fix that to be honest - but most stuff works a lot better now

myxor, 2022-01-19:
It looks like this now:

[image]

Definitively a lot better as we don't have complete black areas anymore!

gfwilliams, 2022-01-19:
Hmm, still not great. I'd added the ability to completely pull out the dark greys but then I found that the text wasn't displaying properly so I had to move the threshold back to ~50%

I think to get it better we might have to look at the surrounding area to choose where we put that threshold value. The actual text seems to be surrounded by a light halo which should help with that

gfwilliams, 2024-05-02:
Fixed now - sadly opentopomap wasn't working for us (often tiles never loaded) so we had to move back to standard openstmap without contours.

A lot of work has gone into filtering though and it's miles better now

closed 2024-05-02 by gfwilliams