espruino/BangleApps #1289: Swiper Clock Launch "return" functional again

elrod16, 2022-01-13:
After installing firmware 2v11.23, Swiper Clock Launch's swipe to return to clock started working again. However, it does not have a blacklist or control over which swipe handlers you want active, while the new Swipe to Clock app implements this functionality better.

thyttan, 2022-11-28:
Candidate for being closed - leave a comment if you object!

Rationale:
This seems to report an absence of an issue, which shouldn't be an issue :P

There are some feature requests here, but those should probably be in a separate issue instead.

gfwilliams, 2022-11-30:
Thanks - yes, I think this can be closed.

`swiperclocklaunch` appears to have been updated recently and so I assume isn't completely broken now.  `swp2clk` has a blacklist in it and maybe could be better, but I guess we have both and people can choose

closed 2022-11-30 by gfwilliams