espruino/BangleApps #1294: messages app: Feature request - variable font size

lightnin, 2022-01-13:
Font size of messages app is a bit small for my old eyes. 

Please consider:

1. Variable font size: When text is < n characters, use larger font size that makes most of screen real estate. > n use a minimum size (that is larger than the one now used) and scroll.
 or
2. Setting to control font size of messages.
and/ or
3. Autoscroll lines of larger text.

It seems like an option to get rid  of the buttons on the bottom in favor of a gesture, in order to retain more usable screen real estate for message text, could be another way to make messages more readable.

Thanks for making this great platform! It's really fantastic.

DDDanny, 2022-01-13:
I support this request.  👍 

I thought in the morning about to use bangle2 touchscreen in showMessage function:
* use the swipe up/down (maybe on the right screen side like clockmode) to imitate something like a scroll/paginate through larger messages
* and furthermore to use also swipes and/or double taps to replace the buttins confirm, abort, back - which also will give the opportunity to have a larger textarea or to support buttom left/right widgets. (Haven't seen one yet)

Any feedback which swipe/tap/drag would fit the best?
I can imagine: 
* back -> swipe left
* confirm -> swipe right (like a lot of "delete" actions in smartphones)
* double tap to abort

During such an extension it should be "easy" to implement small, medium, large font size

DDDanny, 2022-01-13:
prototype swiping (will allow a larger textarea and prepares to set font large for body) without buttons but a global variable and one swipe handler and one duble tap handler

Testmessages - Mock
```javascript
var MESSAGES = [{t:"notify",id:"mail", new: true, title:"This is long title. Very Long. Ipse lorum actum ractum dactum benedictum benedactum showMessage should be scroll :-p", body:"This is long BODY. Very Long. Ipse lorum actum ractum dactum benedictum benedactum showMessage should be scrollalble - yes, scroll and use the touch screen and replace back button with swipe left, so we can read until here with font medium :-p", src:"Mail", subject: "SUBJECT", positive: true, negative: true},
{t:"notify",id:"phone", new: true, title:"short title.", body:"This is short  BODY.", src:"Mail", subject: "SUBJECT"}];
```

```javascript
var currentShowMsgId;
function showMessage(msgid) {
```
....
```javascript
  // Normal text message display
  currentShowMsgId=msgid;
```
No Buttons: so commented out
```
  /*
  var buttons = [
...
if (msg.positive) {
...
if (msg.negative) {
    }});
...
  }*/
```
register touch-handlers that will execute same code via msgid
```javascript
  Bangle.on('tap', function(tapped) { //double tap for back
    console.log("tapped", tapped);
    if (currentShowMsgId && tapped.double) {
      msg=MESSAGES.find(m=>m.id==msgid);
      if (!msg) return;
      msg.new = false; 
      saveMessages(); // read mail
      cancelReloadTimeout(); // don't auto-reload to clock now
      currentShowMsgId=undefined;
      checkMessages({clockIfNoMsg:1,clockIfAllRead:0,showMsgIfUnread:0});
    } // back
  });
  Bangle.on('swipe', function(swiped) {
    console.log("swiped", swiped, "msgid", currentShowMsgId);
    if (currentShowMsgId && (swiped===1 || swiped===-1)) { //swipe right for positive //swipe left for negative
      msg=MESSAGES.find(m=>m.id==currentShowMsgId);
      console.log(msg);
      if (msg && (msg.positive||msg.negative)) {
        msg.new = false; 
        saveMessages();
        cancelReloadTimeout(); // don't auto-reload to clock now
        if (Bangle.messageResponse) //usefull if app is missing or in IDE
          //when swiped===-1 && msg.negative -> send false
          Bangle.messageResponse(msg,swiped===1 && msg.positive); 
        currentShowMsgId=undefined; //reset global, dirty?
        checkMessages({clockIfNoMsg:1,clockIfAllRead:1,showMsgIfUnread:1});
      }
    }
  });  
```
remove buttons from layout:
```javascript
    {type:"txt", font:bodyCfg.font, label:body, fillx:1, filly:1, pad:2 },
    //{type:"h",fillx:1, c: buttons}
  ]});
```
1. I can't get `Bangle.on('tap', function(tapped)` to be fired in the webIDE ?!
2. the swiping works

gfwilliams, 2022-01-14:
Hi, I think the message font sizes have already been mentioned in https://github.com/espruino/BangleApps/issues/1145 / https://github.com/espruino/BangleApps/issues/1148 / https://github.com/espruino/BangleApps/issues/1031

Right now the message title font does change dynamically depending on how much text there is, so doing the same for the body text is probably no big deal.

For the swipe - sounds like a plan *but* if the message is just displayed with no buttons and there's no documentation onscreen nobody will know how to use it. Also how do you know whether you even *can* ack/nack a message as some can't (esp on iOS). It'll need icons right at the bottom I guess to indicate that swiping does a specific action.

@DDDanny these kind of question are best asked in the forum, and I think the reference has info too. But: gesture: needs the gestures file installed and uses the accelerometer, stroke: works for me, but there is a minimum stroke length. I think what you're after is `swipe`?

Also `tap` uses the accelerometer, so again not what you want.