espruino/BangleApps #1305: [github] use Issue and Feature Request-templates

DDDanny, 2022-01-15:
I suggest to start using the bug and feature request templates to better group, bundle issues and also to better cluster per app etc.

Include in the title the development name.

Example in my repro:
https://github.com/DDDanny/BangleApps/tree/master/.github/ISSUE_TEMPLATE

You can create test! issues there ;)

gfwilliams, 2022-01-17:
This sounds like a great idea - thanks! It'd definitely make things a lot more easier, and it's something I'd like to add to the main Espruino repo as well.

Is it possible to add some message (ideally not part of the issue!) along the lines of: "does your issue not fit in here? ask on the forum"

DDDanny, 2022-01-17:
I just knew the old simple templates, but, yes, indeed! 

By using the new [docs.github BETA feature of custom yaml forms](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/syntax-for-issue-forms)

So multiple structured forms for 
- category: issues/bug
    - general bug
    - specific to "Bangle app"
    - ...
- feature request
    - specific to "Bangle app"
    - ...
are possible.

Furthermore it is also possible make a form for pullrequests, if this is helpfull (e.g. agree that included sources are MIT licensed *g*)

Here is some kind of adapted template that has nearly all the features.
A dropdown menu is posible, but that then probably all
There can be a header / hint to the forum within the form.

---
Link: https://github.com/DDDanny/BangleApps/issues/new?assignees=&labels=bug&template=bangle-bug-report-custom-form.yaml&title=%5Bgeneral%2Fapp%2Fwidget%2Fclock+e.g.+clock%5D+%5Bdevelopment+object+e.g.+antonclock%5D+title
---
Hardcopy

[image]

hughbarney, 2022-01-17:
Will this produce 2 lists or does it just force the title into a certain format.
For example  APP:   xxxx,  BANGLE:  xxxxx  FEATURE  xxxxx
Definitely the way to go.

DDDanny, 2022-01-17:
No, all templates run to on the same "Issues" list.
Labels can be defined by the owner (defaults are bug and enhancement) - can also be bug and feature request etc.

Example: https://github.com/DDDanny/BangleApps/issues

I created one issue with bug report template and one with the feature request.
I think it would be good to start with two simple forms (with the link to the forum included)
hw version (must have) and firmware version (optional?) is maybe really good structured info.
When using checkboxes github shows tasks, so my guess is to go with a DDLB Bangle 1, Bangle 2, both and maybe extend in future...

The complicated caption with the brackets is then also no more required, simply clone the bug template and have labels app, widget, clock

There is also the possiblity to open an unstructured issue 

> Don’t see your issue here? Open a blank issue.

gfwilliams, 2022-01-18:
This looks great - HW and FW versions would both be great to have as I've had a lot of issues from folks that never updated firmware.

But maybe just have one big unformatted area everything else with some prompts in? If folks aren't getting it we can always change it later.

I wouldn't worry about the format of a PR at the moment though.

DDDanny, 2022-01-20:
> But maybe just have one big unformatted area everything else with some prompts in? If folks aren't getting it we can always change it later.

Did I understand correctly that :
1. You expect a hint to the forum
2. One textarea prefilled with the steps e.g. What is your hard- and firmware; What is the bug? What is expected; How to reproduce

or

1. You expect a hint to the forum
2. DDLB-Field for HW (Bangle 1, Bangle 2, both)
3. Input-field for FW Version with the link to the "How to update"
4. One textarea prefilled: What is the bug? What is expected; How to reproduce

I hope I can setup two templates latest by Sunday because I have a strong learning curve and workload @work regarding VS Code/Codium with MS Azure and Terraform...

> I wouldn't worry about the format of a PR at the moment though.

Allright, I just wanted to mention that there is the possibility to have this kind of option. You can keep in mind and think about whenever you think "WTF why isn't XYZ mentioned in this PR"

gfwilliams, 2022-01-21:
The second option would be preferable I think. Thanks for this! It should be a big help!

DDDanny, 2022-01-24:
Please check the bug form:
https://github.com/DDDanny/BangleApps/issues/new?assignees=&labels=bug&template=bangle-bug-report-custom-form.yaml&title=Short+description+and+provide+the+affected+app%2Fwidget

remark: I was not able to see my "current fwversion". I would like to place an instruction how to see it...
maybe it should be an option e.g. settings-app -> system - or I should better ask this question at the forum *smile*

Preview:
[image]

gfwilliams, 2022-02-03:
Sorry for the delay - looks great - thanks! Please could you do a PR? I'd have merged it myself but it looks like you're developing some other stuff in the repo.

Luckily the App Loader now shows the firmware version when connected under 'More...'. Otherwise you can check from the About app but not everyone has that

DDDanny, 2022-02-04:
no problem for delay, you do a very well one man show! and give good hints at forum 👍 
yeah i have a lot of open work in my repro - touched a lot apps...

- [ ] I can provide a similiar template for a feature request

closed 2022-02-04 by gfwilliams

gfwilliams, 2022-02-04:
Thanks! Now I've got the format of it I can probably save you some time and tweak that one myself to add a feature request though

PR #1390 (2022-02-04): Git issue template
fixes #1305 part 1; issue template

latest fw 2v12 set in placeholder and inserted the info to the app loader, about app and firmware widget