espruino/BangleApps #1308: Calculator decimal

midnight4577, 2022-01-17:
The Calculator app on the Bangle.js 2 seems to have some strange things going on with the decimal place, it does not appear when it should and instead appears before the "-" sign. I expect is has something to do with the font or is a typo.
1/3 returns:
[image]

And 1-2 returns:
[image]

midnight4577, 2022-01-17:
After uploading to RAM from the Espruino web IDE I have noticed more strange behavior, My decimal seems to be working fine now but now the "-" is not displaying.

bobrippling, 2023-11-13:
Sounds like the calculator app will need tweaking

kayvour, 2025-06-01:
hi, is this still open? I would like to work on this

kayvour, 2025-06-02:
It does work fine here as well, cheers

thyttan, 2025-06-02:
Thanks for checking!