espruino/BangleApps #1319: configurable_clock has no JSON in apps.json

gfwilliams, 2022-01-19:
@rozek  I've just merged the PR from http://forum.espruino.com/conversations/372018

And I noticed that configurable_clock never made it into apps.json. Was that the intention?

For now I'm renaming it to _example_configurable_clock so it doesn't break things, but if it wasn't then please could you put a `metadata.json` file in the app's folder and rename it back?

gfwilliams, 2022-01-19:
Actually it's possible this is due to a merge error... Hang on - i'll check

rozek, 2022-01-19:
Perhaps, it has been deleted by accident.

I'll try to provide the metadata a.s.a.p.

gfwilliams, 2022-01-19:
Thanks! Just checked and I can't see it in the apps.json before the big merge, so if you were able to grab it that'd be great

rozek, 2022-01-19:
Well, I just added the metadata.json (and removed a wrong .md file)

But what I do not understand is: "could you put a metadata.json file in the app's folder and rename it back?" What will I have to rename back?

gfwilliams, 2022-01-19:
Thanks!

I'd moved `apps/configurable_clock` to `testing/configurable_clock` but may not have managed to push it straight away. Anyway, all moved back and fixed now

closed 2022-01-19 by gfwilliams