espruino/BangleApps #1361: Temperature...

zwiglm, 2022-01-28:
It seems that the temp sensor is not working in 0.14

myxor, 2022-01-28:
Which firmware version are you using? 
There was a bug causing the barometer to return no correct values. This is fixed in recent firmware builds.

zwiglm, 2022-01-28:
Using 2v11 release

Which firmware version are you using?
There was a bug causing the barometer to return no correct values. This is 
fixed in recent firmware builds.—

zwiglm, 2022-01-28:
It worked with the last version of LCARS

gfwilliams, 2022-01-28:
So the actual issue is that it's not working in LCARS? Did you try any other apps to check temperature?

HilmarSt, 2022-01-28:
> It seems that the temp sensor is not working in 0.14

Which sensor do you mean?
AFAIK there a two of them, the SoC temperature and the barometer temperature.
The barometer temperature can be checked with the SensiBLE App (until it crashes ;)

gfwilliams, 2022-02-02:
Any update @zwiglm ?

zwiglm, 2022-02-03:
Sorry. Nothing new over here. Maybe the temp-sensor has gone blank. Don't know. (Thought I never dropped the watch or anything other harmful)

Ronin0000, 2022-02-04:
I have had the same Issue on the LCARS Clock, it just shows a dash - .

gfwilliams, 2022-02-04:
So is it just an issue with LCARS? What about one of the thermometer apps? https://banglejs.com/apps/#thermom

Ronin0000, 2022-02-04:
Yes, the temperature app works.

HilmarSt, 2022-02-04:
There a two temperatures in LCARS:
1. the internal (core) temperature of the watch:
`E.getTemperature()`
2. the temperature from the weather app:
```
function getWeather(){
  var weather;
  try {
    weather = require('weather').get();
  } catch(ex) {
    // Return default
  }
  if (weather === undefined){
    weather = {
      temp: "-",
      hum: "-",
      txt: "-",
      wind: "-",
      wdir: "-",
      wrose: "-"
    };
```
The second one returns "-" for the temperature if no weather is installed.

gfwilliams, 2022-02-07:
Ok, so E.getTemperature() should never fail. I think this is likely that you don't have the Gadgetbridge weather stuff set up and we can close this?

peerdavid, 2022-02-11:
As already mentioned - you can select two different temperatures: (1) The core temperature (CoreT in settings) which is loaded using E.getTemparture() as @gfwilliams  mentioned this should not fail and also should not show a "-". I guess if a "-" is shown you selected "Temp" in the settings. So data is loaded from the weather app. Please ensure that the data is transferred from Gadgetbridge - simply open the weather app on your Bangle and check if weather is shown there...

I hope this helps :) Otherwise let me know I'm happy to help :)

gfwilliams, 2022-02-11:
Thanks @peerdavid!