espruino/BangleApps #1362: messages app: unneeded "no messages" notifications from WhatsApp Web

alessandrococco, 2022-01-28:
Hello!
I noticed a small issue with Messages app.

When I'm chatting via WhatsApp Web (PC) every time I receive a message in the _current_ chat I also receive a notification on the Bangle but it says "no messages". I think that the watch should not receive/not display these empty notifications.

gfwilliams, 2022-01-28:
What are you using this with, Android or iPhone? Please could you post a picture? I can't really imagine what this looks like

alessandrococco, 2022-01-28:
My bad, I forgot to fill in the details here!

I'm using an Android 11 smartphone (POCO X3 NFC).

After I receive a message in an active chat I see this:

[image]

I only realized now that this is the default screen of the Messages app. 

It seems that the message is marked as read but the notification is sent anyway. I kept an eye on the watch during a chat: after the buzz I see "loading" and the "no messages" screen appears.

Hope it's clearer now :-)

gfwilliams, 2022-01-28:
Ok, thanks for the info! yes, that's really interesting.

All the apps are up to date? Because the new messages app is supposed to go back to the clock if a message arrives and is then deleted

alessandrococco, 2022-01-28:
Yes, all apps are up to date.  
Moreover I'm on the latest **stable** firmware and I'm using the latest f-droid gadgetbrige.

alessandrococco, 2022-01-31:
Another thing. When I receive a notification the display remains off - is this correct? But when I receive the _no messages_ "notification" the display turns on 🤔

alessandrococco, 2022-02-16:
I uploaded a video showing the issue.

https://user-images.githubusercontent.com/2331798/154351137-dbf07ea1-988c-4bb4-b9db-50c53c8c1f64.mp4

alessandrococco, 2022-04-27:
Some more details!

I can replicate the issue using the Web IDE and **sending two messages within a very short period of time**, e.g. `GB({ "t": "notify", "id": 1575479849, "src": "WhatsApp", "title": "The Nameless",  "body": "CIAO!"}); GB({ "t": "notify-", "id": 1575479849});`

So in my tests I noticed that 

1. the first notification call pushMessage(notify)
2. the second notification call pushMessage(notify-)
3. the `messages.json` is updated (add message, remove message)
4. the first notification call load("messages.app.js") 
5. A kill event is emitted
6. the Messages app is loaded... but the message has been removed so "no messages" is shown

My idea is to wait some instants before pushing the messages (I'd keep them in a tmp array) and then remove a message if the app receives both `notify` and `notify-` for the same id. 

@gfwilliams Does that make sense?

gfwilliams, 2022-04-27:
Excellent, thanks! This helps no end!

I've just pushed a fix to the development app loader. Luckily there was already a delay introduced before the messages app is loaded, so all that was needed was to remove the timeout if the message was removed in such a way that no new messages existed.

alessandrococco, 2022-04-27:
Great, thanks!

commit b7f00b1e (2022-04-27): 0.36: Ensure a new message plus an almost immediate deletion of that message doesn't load the messages app (fix #1362)