espruino/BangleApps #1366: heart tracking app: graphic glitch in the widgets bar

alessandrococco, 2022-01-28:
I just started the HRM (3 min) and the app shows a loooong bar nevertheless there's no data yet.

[image]

After returning to the menu the top of the bar remains in the widgets bar:

[image]

Android 11, latest stable firmware, all apps are up to date.

alessandrococco, 2022-01-28:
I took a look at the code - Would it make sense to remove the `drawWidgets` before drawing the bar chart?

https://github.com/espruino/BangleApps/blob/2c9e164a060d92fc751c551922bf34b01f1a5b4d/apps/health/app.js#L104 

I'd like to try to make the change:
- remove the drawWidgets() calls
- update the code to go fullscreen (steps charts also)

What do you think @gfwilliams? :-)

gfwilliams, 2022-01-31:
It's a bit harder than just removing the drawWidgets call I'm afraid since even if you clear the screen any event that'd cause a widget to redraw (eg battery going down slightly) would put it back on the screen.

Looks like the graph drawing just needs fixing to not draw past the top of the screen - Maybe it should use Bangle.appRect to get the area it can draw in

Ronin0000, 2022-02-08:
Do you just want the bar not to be drawn at all because there is no data or do you want it to draw a long bar in the middle, but not drawing into the widget area?

Ronin0000, 2022-02-08:
I have made a pull request for this.

alessandrococco, 2022-02-08:
> Do you just want the bar not to be drawn at all because there is no data or do you want it to draw a long bar in the middle, but not drawing into the widget area?

Two things:

1. If there is no data for a day/hour then the app should draw nothing for that day/hour
1. The app should not draw into the widget area

Thank you :)

PR #1842 (2022-05-18): [Health] Fix #1366
This PR fixes #1366 (there was a division by zero). It also restores the var I replaced with let some weeks ago.
commit a2b91d94 (2022-05-18): Merge pull request #1842 from alessandrococco/1366-health

[Health] Fix #1366
closed 2022-05-18 by gfwilliams