espruino/BangleApps #1373: Idea: Upload A-GPS data to Bangle.js on connect to Web UI

myxor, 2022-01-31:
Similar to the "Always update time when we connect" option on the Web UI we could have the same for uploading A-GPS data to the Bangle.js.
We would have to configure the timespan (for B1) and providers (for B2) but this should be possible.

What do you think?

gfwilliams, 2022-01-31:
Yes, it could be an idea I guess?

Just to add that if you're on Android, this is some functionality I'd like to add to the Gadgetbridge Android app anyway so it happens in the background

Darin755, 2022-06-17:
> Yes, it could be an idea I guess?
> 
> Just to add that if you're on Android, this is some functionality I'd like to add to the Gadgetbridge Android app anyway so it happens in the background

I would love to see this. I always forget to upload the agps data.

myxor, 2023-03-01:
I guess we can close this as we have multiple ways for automatic A-GPS data update on the watch, right?

gfwilliams, 2023-03-01:
Yes, I think so - thanks!