espruino/BangleApps #1380: WrapString doesn't work with words longer than the given maxWidth

Rarder44, 2022-02-01:
the wrapString doesn't break the single word if is longer than the specified maximum size.

Code:
> var str="VeryLongStringSTRING VeryLongStringSTRING";
> g.setFont("Vector",18);
> var strs=g.wrapString (str,g.getWidth()-2);
> print(JSON.stringify(strs));
> 
> g.clear()
> var h=g.getFontHeight()+1;
> var y=2;
> 
> for( var i =0;i<strs.length;i++)
> {
>   g.drawString(strs[i],2,y);
>   y+=h;
> }

Expected:
a "MS Word-style" wrapString
[image]

gfwilliams, 2022-02-02:
Thanks - this is something that I think we'd have to change in Espruino now.

Is this an issue you've hit in real life, or is it just something you noticed?

Rarder44, 2022-02-02:
I re-implemented the "messages" app; in the changes I have slightly increased the size of the font, and it happens (especially in the titles of the notifications) that long text is not cut on 2 lines but exits the screen.

gfwilliams, 2022-02-11:
This is fixed in new firmwares

closed 2022-02-11 by gfwilliams

Rarder44, 2022-05-09:
i think i found a "bug".
in 2v13 firmware the solution works only for the first line.

i tried to run this code on the clock with 2v13 and the first line is correctly split, while the second continues off the screen.
It appears that if it does not find a space, the line is not split.

`var s=JSON.stringify({"VeryLongTextWithoutSpace":"VeryLongTextWithoutSpace",1:2,"id":10,"name":"marco","surname":"polo"});
g.reset().clear(Bangle.appRect);
g.setFont("6x8").setFontAlign(-1,0);
s = g.wrapString(s,g.getWidth()-8);
g.drawString(s.join("\n"), 4, g.getHeight()/2);`

gfwilliams, 2022-05-10:
You mean the issue is if you have a really long word that is more than double the screen width, it won't keep splitting it?

So `VeryLongTextWithoutSpace` only goes to `Very,LongTextWithoutSpace` not `Very,Long,Text,With,outS,pace`?

Rarder44, 2022-05-10:
exactly

gfwilliams, 2022-05-10:
Ok, thanks. I guess in most normal cases this doesn't happen - just filed an issue for it on https://github.com/espruino/Espruino/issues/2194 - probably not going to end up super high priority for me though