espruino/BangleApps #1383: App Loader "Update App Button" does not work on customizable apps.

Ronin0000, 2022-02-02:
When I try to use the Update Apps Button on languages, which is a customizable app than needs to be updated, it says this error:
[image]
When I press the update button on the languages app it works fine and opens the customization window.
[image]

gfwilliams, 2022-02-03:
That's odd - I though we'd fixed that so the 'Update app' button didn't include customizable apps.

But the idea is that the 'Update app' button does not update custom apps. It used to AFAIK and someone filed an issue saying not to - it's not great user experience having a bunch of custom app windows popping up if you click 'Update'.

bernhardreiter, 2022-03-12:
I am seeing the same issue, with the language app installed and customized for German.

Suggestion: If it cannot be fixed directly, the message could be changed to say something like `Updated failed, no apps can be updated automatically.  (Some apps require a manual update with configuration.)`

closed 2022-03-30 by gfwilliams