espruino/BangleApps #1405: messages app: Message display autoscroll to display more of message body if larger than single screen.

lightnin, 2022-02-07:
When a message body is longer than what can be displayed on screen at one time at the selected font-size, it should be automatically and slowly scrolled upwards after a slight pause, such that however much of it fits into 3 times the visible screen area used at a given font-size can be displayed. If larger than 3x total screen area, truncate message and add an elipsis at the end.

Timeline of animation:
1. Display top visible portion of message
If message # lines > display area # lines:
2. at 1.5 seconds, begin scrolling the message vertically, approx. .75 lines per second.
3. When bottom of message is visible, stop scrolling and pause for 1 second.
4. Go to (1) and repeat while message is actively being displayed.

Notes: 
Implementation should work regardless of selected font size in order to be compatible with possible future font size settings.

lightnin, 2022-02-07:
It looks like this can be done by rendering the body text as a rectangle, and when the rectangle is vertically larger than can fit in the visible space, it should be cropped at progressively lower points before being fed to the layout and rendered. But my JS-fu isn't yet strong enough to see how best to keep track of vertical position between renders, and how to manage the timing in a way that doesn't block or waste battery.

gfwilliams, 2022-02-08:
I'm not sure many people would appreciate this - it'll either be too fast or too slow.

There's definitely been talk of drag to scroll though - probably a tap on a message that's too long to take you to a new page where you get the whole screen area available to show you the message you received.

commit 83a849a9 (2022-02-08): 0.20: If message doesn't fit, allow tapping on the body to show a scrollable
      view of the message and title in a bigger font (fix #1405, #1031)
closed 2022-02-08 by gfwilliams