espruino/BangleApps #1418: Configurable Analog Clock: Does not load

nxdefiant, 2022-02-10:
Hi,

I tried the Configurable Analog Clock on BangleJS2 and the screen gets stuck on "Loading...". In the Browser console I see 

> Uncaught Error: Module https://raw.githubusercontent.com/rozek/banglejs-2-simpled-clock-hands/main/ClockHands.js not found
 at line 10 col 351 in ac_ac.app.js
...k-hands/main/ClockHands.js'),complications:{tl:undefined,t:require('...

johan-m-o, 2022-02-11:
Looks like a typo... It's supposed to be "simple" clock hands, not "simpled". @rozek

rozek, 2022-02-11:
Thank you very much for your hints!

I've fixed the typo and changed the version number to 1.0.

A pull request is pending. After a successful merge, the new (corrected) version will be available in the official app loader - just watch for version 1.0!

nxdefiant, 2022-02-11:
That was quick, thanks

rozek, 2022-02-11:
well, here we say "Impossible things are done immediately, miracles take a little longer"

rozek, 2022-02-11:
the pull request for my bug fix has just been merged

closed 2022-02-11 by gfwilliams