espruino/BangleApps #1444: [OpenStreetMap] OSM app always plots the currently select GPS Recorder track

dukky, 2022-02-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

[This commit](https://github.com/espruino/BangleApps/commit/947731a129fdb72281a69556df33c246b4d3280a) introduced the feature for the OpenStreetMap app to plot the currently active GPS track on loading the OSM app. 

The intention for this seems to have been so that while recording a track, you could see it update in real time on the OSM app, which I agree is a nice feature though perhaps it could be controlled in either than OSM or GPS Recorder

However, it also draws the currently selected track even when not recording, which means if a long track is selected, the app is unresponsive until the track finishes drawing. 

With a track that has only 53 records, this isn't really an issue as the draw completes nearly instantly (aside from the track appearing when I don't specifically want it to), but on a longer track (77 mins, 927 records) this takes ~8 seconds, and the app is completely unresponsive until the draw completes, and it redraws on every move of the map.

I think the OpenStreetMap app either needs to just detect that the GPS recorder app is recording (currently it just checks `WIDGETS["gpsrec"] && WIDGETS["gpsrec"].plotTrack`, and the widget is always there with a `plotTrack` function even when not recording)

nxdefiant, 2022-05-29:
I guess the newer [recorder app](https://banglejs.com/apps/?id=recorder) does not have this problem.

gfwilliams, 2022-06-06:
That is a point. It probably shouldn't be checking for `gpsrec` now anyway. I think this is probably functionality that users would appreciate (albeit for `recorder`), but yes, I guess it should be switchable

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
Done with https://github.com/espruino/BangleApps/commit/eba77c8a9915180d1b822b04c292b7f68a4952fd.

closed 2022-12-05 by gfwilliams