espruino/BangleApps #1445: [sleeplog] Having sleeplog enabled hinders execution of kill event handlers

halemmerich, 2022-02-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

For reproduction install sleeplog and enable it. Then run the following code from IDE:
```
E.on('kill',()=>print("killed"));
```
Cause a reload, "killed" will not be printed.
Then deinstall sleeplog and run the code again.
A reload prints "killed"

storm64, 2022-02-13:
I'm just working on a new version with a power saving mode and therefore removed the unnecessary `global.` prefixes, but this bug is the same on my version.

I tracked down the bug to the following code line inside the `global.sleeplog.stop()` function:
```
E.removeListener('kill', sleeplog.stop);
```
Changing the event listener to be called through an arrow function prevents this issue:
```
E.removeListener('kill', () => sleeplog.stop());
```
 
I will change this in my upcoming version. Just doing a last overnight test before releasing tomorrow.

---

This fixes the issue but I'm still trying to find the cause.

I assumed the problem occurs when the system is executing `E.removeListener('kill', global.sleeplog.stop);`. The value of `global.sleeplog.stop` might be set to _undefined_  already.

But testing this assumption by printing its value via the following code, seems to prove me wrong? 
```
>E.on('kill',()=>print("global.sleeplog.stop:", global.sleeplog.stop));
=undefined
>load()
=undefined
global.sleeplog.stop: function () {
  ...
}
>
```

gfwilliams, 2022-02-14:
It seems to me that `E.removeListener('kill', () => sleeplog.stop());` won't do anything, because the function you're removing has just been created. I just tried some tests along the lines of:

```
x = () => print("Hello")
y = () => print("World")
E.on('kill',x)
E.on('kill',y)
E.removeListener('kill',x)
```

But it all still works for me...

Please can you print the value returned by `E["#onkill"]` and see what's in it? That is the list of currently set handlers for the `kill` event

gfwilliams, 2022-02-14:
So you're saying that the onkill handler does actually seem to be called? Or is it that the onKill handler stops executing when you call `E.removeListener('kill', sleeplog.stop);` from within it?

gfwilliams, 2022-02-14:
It's possible that some error occurs during execution which stops it running. I guess in that case, because the event is being executed *just* before the Bangle rips everything down, the error may not be reported?

gfwilliams, 2022-02-14:
... sorry for all the comments. So that's the issue I think:

```
E.on('kill', ()=>{throw new Error("Testing")})
```

Doesn't show anything at all. I'll add something to Espruino so it checks for errors and reports them,

storm64, 2022-02-14:
I just installed the 2v12.32 firmware and reproduced the circumstances first mentioned by halemmerich.
On a reload where now an error should occur nothing is shown on the console.

Detailed setup:
1) Replaced both `E.on('kill', () => sleeplog.stop());` and `E.removeListener('kill', () => sleeplog.stop());`
with `E.on('kill', sleeplog.stop);` and `E.removeListener('kill', sleeplog.stop);` to make the issue occur again.
2) Forced the watch to reload with `load();` twice. (First time to activate the new boot code, second to check if a error is thrown.) No error is written to the console.
3) Recheck if the issue still is present by adding a test kill listener as mentioned: `E.on('kill',()=>print("killed"));` and reloading afterwards. Still no error occurs and "killed" is not printed.

I'm not deep enough in the topic to understand what exactly is happening, but it seems that the execution of `E.removeListener('kill', sleeplog.stop);` inside its function deletes all other kill event listeners, whereas executing `E.removeListener('kill', () => sleeplog.stop());` does not cause any trouble.

gfwilliams, 2022-02-15:
Ok, thanks. It seems that rearranging the list of events while executing it ends up messing up execution. Just tested and can reproduce with:

```
function test() {
  print("A");
  E.removeListener("foo",test);
  print("B");
}
E.on('foo', test);
E.on('foo', () => print(c));
E.emit("foo");
```

To fix this properly on existing firmwares, please could you do something like:

```
function stopHandler() {
  // your original code here
}

sleeplog.stop = function() {
  E.removeListener('kill', stopHandler);
  stopHandler();
}

E.on('kill', stopHandler);
```

So now you call `sleeplog.stop` from your code as you did before, but when the 'kill' event occurs you don't try and remove the handler while it's running.

storm64, 2022-02-15:
I already used the following approach that seems to work quiet well to fix the issue (see pull request #1448):
```
function test() {
  print("A");
  E.removeListener("foo", () => test());
  print("B");
}
E.on('foo', () => test());
E.on('foo', () => print("C"));
E.emit("foo");
```
Returns: ` A `, ` B ` and ` C `
Do you think this needs to be changed?

---

While fiddling around with the code I experienced the following behavior:
```
function test() {
  print("A");
  E.removeListener("foo", test);
  print("B");
}
E.on('foo', test);
E.on('foo', () => print("C"));
E.emit("foo");
```
Returns: ` A ` and ` B `
Continue with:
```
E.emit("foo");
```
Returns: ` C `

The second emit seems to work as expected: The test function is removed and the reminding listener is called.

gfwilliams, 2022-02-15:
Yes, it needs changing - as I'd mentioned above `E.on('kill', () => sleeplog.stop());` / `E.removeListener('kill', () => sleeplog.stop());` doesn't work - it won't remove the listener at all.

commit 7b72606a (2022-02-14): Merge pull request #1448 from storm64/master

sleeplog: Add power saving mode, fix #1445
commit 8c4f2f18 (2022-02-16): sleeplog: Correct fix #1445, display loading info while calculating sleep data

Update app.js
 - move complex data analysis into separate function
 - add displaying a loading info
 - faster feedback by calling the analysis through a timeout
Update boot.js
 - correct fix #1445 as mentioned by splitting the `stop()` function to only remove the kill listener on a manual stop
 - optimize `start()` and `stop()` functions
Update README.md
 - change beautified menu entries according to changes in settings.js
Update settings.js
 - replace `circulate()` function with internal `wrap: true`
 - move delay [... cut]
commit b896d5d0 (2022-02-16): Merge pull request #1459 from storm64/master

sleeplog: Fix #1445, display loading info, add icons to display service states