espruino/BangleApps #1455: [Health Tracker] Heart rate monitoring sometimes turns screen backlight on (?)

jes, 2022-02-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

I turned on heart rate tracking in the health tracking app, and it seems to cause the watch's backlight to turn on occasionally. This is very annoying at night time.

It looks like the screen turning on is sync'd with the 3 minute period, but I haven't observed it enough times to be sure. It also appears to have stopped happening since I turned off heart rate tracking, which is why I think it is caused by heart rate tracking.

I have not been able to reproduce it with `Bangle.setHRMPower(1)` in the console, and I've also had the watch running now with heart rate tracking switched back on for the last 10 minutes and haven't seen the screen turn on, so I don't think it happens every time, but I don't know what else could be making the backlight come on in the middle of the night.

gfwilliams, 2022-02-15:
Could it be the bluetooth connection dropping in and out? That is sometimes a cause.

It seems extremely unlikely it's heart rate monitoring - I've never seen that happening, and the HRM code internally never touches the screen

jes, 2022-02-15:
Thanks, it could well be! I'll close this on the assumption that it is that, and I'll reopen if I discover otherwise.

closed 2022-02-15 by jes