espruino/BangleApps #1460: [messages] file write optimisation

gfwilliams, 2022-02-16:
Right now, we write a new JSON file every time a message is added or removed. If you're constantly getting notifications this could make Bangle.js run slow pretty quickly.

I think it'd make things a lot faster if we just used StorageFile instead, and we *only* append to it. Either a new message, or something saying a previous message should be removed. We could then check if the StorageFile was getting too big and just delete it and create a new one without the junk in it.

Not too hard, and it should help a lot, at the expense of a bit of a delay when reading it

jeroenpeters1986, 2022-02-17:
I was reading this and I thought I got it, but I think I don't. 
Are you also planning on changing the format, like a row per message?
I am a bit lost on only appending and something saying a previous message should be removed. Arent you then drawing messages and later retracting them?

rigrig, 2022-02-21:
Sounds like a good idea.   

If we're going to change message storage, maybe we should special-case music, as it seems those lead to a lot of pointless storage activity? i.e. keep `musicstate`/`musicinfo` in memory, and only write+read them to flash when launching the app?

gfwilliams, 2022-02-21:
That's a good point - actually it might be worth just special-casing if the messages app is open - storing the messages in RAM and then only writing them on exit. Things like navigation will also really thrash the storage otherwise

rigrig, 2022-04-22:
If we're messing with it anyway, it might be a good idea to decouple the library from the app.

In [my (WIP) message app rewrite](https://github.com/rigrig/BangleApps/blob/new-messages/apps/messages/lib.js) I changed it to do `Bangle.emit("message", type, event);`.
I don't think we want different message apps, but maybe clocks can do something nice with it, and it seems cleaner in any case.

gfwilliams, 2022-07-06:
Sorry for taking so long to check https://github.com/espruino/BangleApps/pull/1498 out properly... I still think this is a good plan - I just wanted to sit down and properly run through what's happening in the PR and haven't felt like I had time yet.

> maybe we could just store messages as a regular file each?

Yes, that would definitely be a big improvement over the current situation.

... however what really slows the Bangles down at the moment is having lots of 'trash' files in Storage. Your solution would use a lot less storage (and RAM) but would still generate one new file per message (however deleting a message wouldn't create a file so I guess it's twice as good as now ;)

The PR should *drastically* improve things though. While StorageFile does just use Storage, it doesn't add one new file per write - instead it pre-allocates a file and then just fills it up, and when full it creates a new file - so actually maybe it's only adding one new file per 500 writes or so.

rigrig, 2022-12-07:
[Looks like](https://github.com/espruino/BangleApps/pull/1498#issuecomment-1340858723) this can be closed now?

gfwilliams, 2022-12-07:
Thanks, yes - replaced this with https://github.com/espruino/BangleApps/issues/2370

bobrippling, 2023-11-13:
Closing as mentioned