espruino/BangleApps #1467: [Feature request Bangle.js 2] Method for retrieving the last date when the battery was charged

HilmarSt, 2022-02-17:
Charging of the Bangle.js 2 is done with a magnetic connector which invalidates any compass calibration.

With the help of a new or an extended old function, a new compass calibration can be requested by an app after proving that the last calibration date is older than the last charging date.

For instance:
- extend <code>E.getBattery()</code> to return the last charging date
- add a new function <code>E.getChargingDate()</code>

bobrippling, 2023-11-13:
I think Gordon meant you could add an app that has boot code which will save the date when it gets a `charging` event, and use that library for when you want the last charged date. Closing this for now but feel free to re-open or create a PR if you like

closed 2023-11-13 by bobrippling

nxdefiant, 2023-11-13:
I believe this is also fixed in #2648