espruino/BangleApps #1477: [timerclk] Cannot remove alarm

f616, 2022-02-18:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

I have a persistent alarm set to everyday at 12:00:00 which cannot be removed and doesn't appear on the alarm list

andrewgoz, 2022-02-25:
I've noticed that Timer Clock defaults to alarms being active. Try pressing the 'X' lots of times to make sure all alarms are deleted, then in the alarm that remains, press the blue button so it looks like a 'play' button (triangle). Then exit out by pressing the button - don't swipe to look at other alarms.

f616, 2022-02-26:
That seems to "solve" the issue. Thank you

Ronin0000, 2022-03-04:
Maybe this should be closed now that the issue is solved

closed 2022-03-04 by gfwilliams