espruino/BangleApps #1482: Feature Req:   Bangle.on('xxxPower') events

hughbarney, 2022-02-19:
It would be very useful to have events for changes in the power status to devices like gps, hrm, compass.
The gps being the most important as it is a major consumer of power.

The reasons for this are discussed in defect ###

gfwilliams, 2022-02-21:
As you mentioned in the linked issue:

> what is needed is a power change event so that the widget gets poked into redisplaying the status

It's actually pretty easy to just override the setGPSPower and this can be done for any other event you care about:

```
myGPSChangeHandler(Bangle.isGPSOn());
(function(on,id) {
  var p = Bangle.setGPSPower;
  var isGPSon = p(on,id);
  myGPSChangeHandler(isGPSon);
  return isGPSon;
})()
```

I seem to recall this was asked in the forum before too.

I don't really see the need for polling, or even really specialist events - since doing the override is pretty quick and there's really not a big performance hit from doing this

hughbarney, 2022-02-21:
I dont really understand what the code above is doing or where you would put such code.

Are you saying that a widget can change the Bangle.setGPSPower() function such that when any other app calls 
Bangle.setGPSPower that myGPSChangeHandler will get called with a value of 1 = powered on 0 = powered off ?

I dont see what is being done below - this seems to be an anonymous block.
(function(on,id) {
  // etc - is this the widget code ?
})()

I could understand if it means something like:

// inside the widget code, override setGPSPower function

function myGPSChangeHandler(p) {
    if (p) {
       print("GPS was powered on");
     } else {
        print("GPS was powered on");
    }
}

// reference old function
var p = Bangle.setGPSPower;

Bangle.setGPSPower = function(on,id) {
   var wasPoweredOn = p(on,id);
   myGPSChangeHandler(wasPoweredOn);
   return wasPoweredOn; 
};

hughbarney, 2022-02-22:
Thanks for clearing that up and updating the example.
I will close this as not required.
This is useful documentation for me at least.

It was not me that added the other colors based on getting a GPS fix.  That needs to be thought out.  Not sure having an interval every 2 seconds or listenning to on('gps') is a good idea in a widget. It might be simpler to revert it to the original idea of power status only.  I do agree with you that everything should be as lightweight as it possibly can be.

closed 2022-02-22 by hughbarney

gfwilliams, 2022-02-23:
Thanks - yes, it's hard to know where something like that might be documented. I guess it's best just to have it used in apps/widgets and then others will find it there.

About the GPS widget - I think showing GPS fix state is a good idea (as it's pretty useful info for GPS) but yes, how it does that might need some thought as it's not immediately obvious from colour. I don't think having an `on('GPS'` handler is something you'd need to worry about too much memory-usage or performance-wise - it could even be added *only* when GPS power is on.