espruino/BangleApps #1483: Feature Req: new apps to be highlighted in the App Loader through tag/menu etc

hughbarney, 2022-02-19:
It would be really useful for new Apps to be detected by the App Loader (somehow). I'm often checking for new Apps and the only way to do this is look at the code coming in from a merge or to painstakingly scroll through all 200+ apps to see if anything new pops out, this can often mean you dont spot things for a while.

alessandrococco, 2022-02-19:
> I'm often checking for new Apps and the only way to do this is look at the code coming in from a merge or to painstakingly scroll through all 200+ apps

There's a sort by new filter:

[image]

myxor, 2022-02-20:
Just want to add that sorting is only available on the stable app loader not on the development one.

gfwilliams, 2022-02-21:
Yes, surely the sort by new is good enough?

hughbarney, 2022-02-21:
> Just want to add that sorting is only available on the stable app loader not on the development one.

So I fetch/merge the latest code into my repo then push the code up to my github loader.  This enables me to try apps out when I see them come in from a fetch.  It sometimes takes several days for an App to come through to the Production loader page.

Not sure why these loader features would not be part of everyones App Loader ?

gfwilliams, 2022-02-22:
> Not sure why these loader features would not be part of everyones App Loader ?

Because they require a script which runs through all the GitHub commits working out which files are newer or not. It's not automatic - you need to run `bin/pre-publish.sh`. My server isn't very fast, but it takes a few minutes to complete on that.

If you want to run that yourself and then commit the result in your repo then everything will work fine and those options will appear.

I think right now it's best to close this. If someone wants to write some magic code which is able to query the GitHub API to find out the dates of all the apps when the app loader is hosted on GitHub pages then a PR would be great though.

closed 2022-02-22 by gfwilliams