espruino/BangleApps #1484: Feature request - Desktop Launcher - make button return to default clock when launch on touch is enabled

hughbarney, 2022-02-19:
I love the Desktop Launcher on Bangle 2 and on Bangle 1 as well.

Sometimes I load it just to look at what apps I have installed OR sometimes it might get triggered by mistake.
When lauch on touch is enabled it would be great if just pressing the button on B2 caused you to exit and load the default clock.

@halemmerich - I thought and implementing but thought I should log this first with you.  I think you might be the original author - am I right ?

halemmerich, 2022-02-21:
I am not the original author, just did some changes.
Since the button currently does nothing in case of direct launch, I like the idea of using it to get back to the clock. It would be better than having to long press every time.

halemmerich, 2022-03-14:
This has been implemented in #1530 by @Rarder44

closed 2022-03-15 by gfwilliams