espruino/BangleApps #1492: Heart Rate Monitor 0.08 - only way to exit is to reboot the watch - losing step count data

hughbarney, 2022-02-20:
### Affected hardware version

Bangle 2

### Your firmware version

2v12.32

### The bug

The only way to exit the Heart Rate Monitor application is to long press reboot the watch.

Reboots now cause all the daily counters to be reset. For example the daily step count will get reset to 0.

A BTN2 press when the watch is unlocked to bring up an END/CONTINUE dialogue would be better.

myxor, 2022-02-20:
I can exit the app by a long press on btn1 without a reboot.

gfwilliams, 2022-02-21:
> I can exit the app by a long press on btn1 without a reboot.

I can too. Please can you try again @hughbarney ? Have you got anything else installed that might be messing this up?

hughbarney, 2022-02-21:
Nothing odd installed. All other apps exit easily on a Bangle 2 BTN press.  A long BTN press reboots the watch on a Bangle 2 and this sets the step count back to 0 - as it you had installed new firmware etc.

There is no Bangle.setUI("clock"); in the code.  Just added and it now exits as I would expect an app to do.

closed 2022-02-21 by hughbarney

gfwilliams, 2022-02-22:
Ok, thanks!