espruino/BangleApps #1511: messages app: Option to not show buttons on alerts

derikb, 2022-02-25:
When I get a new message/alert on the screen the three buttons take up a _lot_ of space, which seems to be why the text is so small.

Might be nice to have an option to not show those buttons, so there'd be more room for the text to be a little more readable.

Without buttons tapping the screen/message could just act as "dismiss". One could always go into the message app itself later if one wanted to do something else, but personally most of the time I just want to see who/what the message was not necessarily do anything with it from the watch.

gfwilliams, 2022-02-28:
There are already a few of threads about this on the forum:

http://forum.espruino.com/conversations/373307/#comment16412381

http://forum.espruino.com/conversations/373174/#comment16410922

I could swear there was an open issue too, but I don't see it now so might as well leave this open

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
It may not be in the default Message UI app, but this is pretty much implemented in Messages Light.