espruino/BangleApps #1516: Feature request: Health: more statistics data

myxor, 2022-02-26:
I would like to collect a list of possible further statistics data which the Health app could deliver in future:

- Total Steps
- Distance (steps * stepLength, stepLength has to be configured as basic body value)
- Min Heart Rate
- Max Heart Rate
- Resting Heart Rate
- Floors Ascended, descended (could we calculate this out of the step sensor?
- KiloCalories (could we calculate this out of some basic body values like height, weight + steps / active time?)

Open for discussion and ideas.

gfwilliams, 2022-02-28:
It's worth thinking about what can be done with the existing recorded data (total steps, distance, kilocalories) and what would require changes to the data that is stored or even firmware changes

gfwilliams, 2025-03-05:
Closing this - moving discussion to https://github.com/orgs/espruino/discussions/7696

closed 2025-03-05 by gfwilliams