espruino/BangleApps #1521: Feature Request: Barometer - Pressure change alerts

fparri, 2022-02-27:
Would that be possibile to add Air Pressure change alerts?

https://www.thoughtco.com/how-to-read-a-barometer-3444043

gfwilliams, 2022-02-28:
You mean in the barometer app itself, or just as something that pops up in normal use?

It's definitely something that'd be pretty easy to do, but with it on your wrist the pressure will change a lot just due to changes in height as you walk up and down stairs, so I guess you'd have to be careful that didn't trigger anything.

fparri, 2022-02-28:
I am testing a Huawei Watch GT Runner: it offers the possibility to enable or disable alerts for pressure drops, measuring the drop in hPA over the last 3 hours, with the following options:

2 hPa/3 h
2.5 hPa/3 h
3 hPa/3 h
3.5 hPa/3 h
4 hPa/3 h
4.5 hPa/3 h
5 hPa/3 h
5.5 hPa/3 h
6 hPa/3 h

I attach some pics

[image]
[image]
[image]

gfwilliams, 2022-02-28:
Looks like a good idea - it could be handled as a widget pretty easily.

Honestly though, I've spent the entire day today *just* answering messages, so for me personally I don't have much time to implement something like this

myxor, 2022-03-05:
I am working on a widget which implements this. Will create a merge request in the next days.

fparri, 2022-03-07:
Just read and just seen your pull request. Looking forward to trying it! :)
Thanks a lot!

myxor, 2022-03-11:
@fparri v0.02 is available since yesterday. Please give it a try and i am happy to receive feedback from you :)

fparri, 2022-03-12:
> @fparri v0.02 is available since yesterday. Please give it a try and i am happy to receive feedback from you :)

Avsolutely. I'm going to update the app and try it tomorrow. Thanks a lot :))))))

fparri, 2022-03-14:
@myxor I tried it yesterday. It worked perfectly. The only difference I noticed with the same feature on my Huawei GT Runner is that the GT Runner probably compensates pressure variation due to a change in altitude.
I got a pressure drop alert on the Bangle.JS 2 but not on the GT Runner. Not a huge deal, though. Thanks a lot again for your app :)

myxor, 2022-03-14:
@fparri how do you think should the compensation Worky? Any ideas?

fparri, 2022-03-14:
@myxor I am not an expert in this area, but I've found various correction tables online:

https://www.engineeringtoolbox.com/barometers-elevation-compensation-d_1812.html
https://novalynx.com/manuals/bp-elevation-correction-tables.pdf
https://airdensityonline.com/2016/11/corrected-and-uncorrected-barometer/

myxor, 2022-03-15:
It could be that such or similar corrections are already included in the firmware of the barometer chip.
Maybe @gfwilliams can confirm this?

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
Done with [Barometer Alarm Widget](https://espruino.github.io/BangleApps/?q=widbaroalarm).

gfwilliams, 2022-12-02:
Yes - looks like [Barometer Alarm Widget](https://espruino.github.io/BangleApps/?q=widbaroalarm) does this

@myxor I don't know about the corrections - you would have to look in the barometer datasheet. There is definitely *some* calculation going on at https://github.com/espruino/Espruino/blob/970089d785ed263f7f9434eba7a3fd8cb9158875/libs/banglejs/jswrap_bangle.c#L4577 though

closed 2022-12-02 by gfwilliams