espruino/BangleApps #1534: Vector Clock v0.07 makes widgets "blink"

StuntsPT, 2022-03-02:
Ever since I have updated Vector Clock on my Bangle.js 1 to version 0.07, widgets started to "blink" every time the seconds change.
Reverting commit e70438d642b6cd6ab8a28afbb1403d11b989e303 seems to fix it, but I'm not sure if it breaks something on the Bangle.js 2.
Is is safe to make a PR with [this line](https://github.com/espruino/BangleApps/blob/master/apps/vectorclock/app.js#L84) removed? Or does it have some unexpected side-effect on the Bangle.js 2?

Thank you in advance.

gfwilliams, 2022-03-02:
It should only be required if the screen is cleared or the widget area gets overwritten. I don't think there's a reason it'd be needed for Bangle.js 2.

Maybe try without for a while and see if there are any problems?

StuntsPT, 2022-03-02:
Alrighty. I've commented that line and uploaded it to my Bangle.js 1 and haven't noticed any issues so far. But I'll give it a week and report back (or earlier if I encounter any issues).
I'm also pinging @HilmarSt since it's his commit. Can you please let us know what issue this line addresses? (Since I might not bump into any problems with my regular use case)

Thank you.

HilmarSt, 2022-03-12:
Fixed with pull request https://github.com/espruino/BangleApps/pull/1563
(Don't know how to do a separate PR for this fix - mabe this only works with a git *clone* and not a *fork* I'm currently using)

nebbishhacker, 2022-03-13:
@HilmarSt may I ask what your reasoning was for redrawing the widgets in the first place? Normally after the initial render the widgets are responsible for redrawing themselves.

StuntsPT, 2022-03-14:
I'm also reporting that after almost 2 weeks of use with the redraw line removed I did not encounter any problems (Bangle.js 1).
I'll be trying out the patch from #1563 tomorrow.

HilmarSt, 2022-03-15:
> @HilmarSt may I ask what your reasoning was for redrawing the widgets in the first place? Normally after the initial render the widgets are responsible for redrawing themselves.

Yes, but many Bangle.js 2 Apps don't set-up a timer to do so - they rely on an redraw when the watch is unlocked. This happens with many Apps that are time or date dependent, e.g. the Days Left Widget, the Moon Phase Widget and some of the battery widgets. It's very annoying to see the widgets only updating after an unlock, some times even a call of the settings menu is necessary.
Maybe this is different on the Bangle.js 1 (that I don't have), also on the Bangle.js 2 there are no blinking widgets if the seconds are updated.
With the fix I've disabled the widget redraw in Vector Clock for Bangle.js 1.

StuntsPT, 2022-03-16:
I've been using the patch from #1563 and is seems to fix the problem on the Bangle.js 1.

> Yes, but many Bangle.js 2 Apps don't set-up a timer to do so - they rely on an redraw when the watch is unlocked. This happens with many Apps that are time or date dependent, e.g. the Days Left [...]

I'm not sure this is an issue with the Bangle.js 2 or with the specific widgets you mention. But in my Bangle.js 1 I can tell that the widgets auto-update themselves (Bluetooth, GPS, memory, and pedometer all seem to change while I'm watching them).

Regardless, I would dare say that widget redrawing should not have to be handled by the watch-face, which would avoid this type of work-around.

gfwilliams, 2022-03-16:
> many Bangle.js 2 Apps don't set-up a timer to do so - they rely on an redraw when the watch is unlocked

Maybe you could let us know which ones so we can fix them? Widgets definitely shouldn't be reliant on the clock face to ask them to redraw - all the ones I know of - bluetooth, gps, time, battery, messages, all update themselves

StuntsPT, 2022-03-24:
I can confirm everything has been in working order after a few days of using v0.08.
I'm not sure on whether I should close the issue or wait for another reply from @HilmarSt for completeness sake (although in fairness, it might warrant a new issue of its own.

gfwilliams, 2022-03-25:
Great, thanks! I think it's fine to close - I'll do it now

closed 2022-03-25 by gfwilliams