espruino/BangleApps #1538: Feature Req: Font converter to have mini Bangle 1/2 emulator to show what the font will look like

hughbarney, 2022-03-03:
The cycle time to try out new font files on the Bangle is quite long and ultimately the only way to see what the font will look like
is to write code to try it out on the actual device.  This requires the time to edit and upload, restart the device etc.  Very time consuming when you are searching for a particular look etc and it you cycle through 10 fonts at a time.   The preview screen at the bottom of the font converter is not terribly helpful as the size is not representative of what the font will look like on the Bangle.

It would be really useful to have a little display that emulated the Bangle 1/2 size and showed what the font will look like on the device.  This might require some of the Emulator to be embedded into the font tool.

If this is not a quick win and requires lots of effort then I guess just pop it on the back burner.

gfwilliams, 2022-03-04:
What did you want to be written on the screen?

It's not like we really need an emulator - we just need to write out the bitmap back onto the screen. For instance the `THIS IS A TEST OF THE FONT` text could just be written with the rendered font rather than using the web browser.

I'd been considering just removing the 100% scale from the font preview box, so it was 'actual-size' - that would probably help a lot.

I'm pretty busy at the moment so this is pretty low priority for me, but it'd be great if someone wanted to jump in and do this

hughbarney, 2022-03-04:
Yes probably does not need the emulator - just a bitmap on a black or white background.

> I'd been considering just removing the 100% scale from the font preview box, so it was 'actual-size'

Agree.  Also remove the grid background and display the text as a string rendered as if it had been painted by g.drawString().  It looks like you are displaying each character seperately inside its own square, which makes it hard to see how the font will look in practice.

Regarding text - maybe grab that from an HTML input and default it to 'The quick brown fox....' . The google font site allows you type a string which is then displayed as the sample for all the fonts.

hughbarney, 2022-11-05:
looks like this has been done so closing.

closed 2022-11-05 by hughbarney