espruino/BangleApps #1542: App appreciation

GrandVizierOlaf, 2022-03-04:
Is there a good way to let a creator know that we are using and enjoying their app? It's happened a few times where I find exactly what I was hoping existed or discover something that I had never considered which improves my use of the watch. When this happens I want to let the creator know that their work is used and let the broader community know that the app is working as advertised.

These immediately come to mind, but there are many others:

- Ratchet Launcher by @jsphpl 
- Sleep Log by @storm64
- BanglExercise by @myxor 
- Many of the bootstrap apps by @gfwilliams

Ronin0000, 2022-03-04:
Nothing like this exists right now but maybe it could be a heart button or https://github.com/espruino/BangleApps/issues/238;

gfwilliams, 2022-03-07:
Thanks! Right now there isn't anything, but I think the reviews in https://github.com/espruino/BangleApps/issues/238 would be the best way to go.

It's been suggested that we could count the amount of people that have favourited an app - we'd just have to be careful about user privacy.

I can see the case where an app might be broken in some way and someone gives it a low score (or doesn't favourite when they might have), so it a way having a description so app developers can see what's wrong would be good (and maybe a way to mark that particular complaint as 'fixed' :).

alessandrococco, 2022-03-07:
A custom Disqus integration could be a start. It supports comments and we could use custom buttons to simulates a rating 🤔 

[image]

hughbarney, 2022-04-01:
Any progress on this.  I'd love to see how my apps are doing.

nxdefiant, 2022-04-30:
Please don't track the downloads. Sure we can use our own app loader to avoid being in the statistics. But how about this simple metric: Number of different contributors to an app. The more authors the more it is likely used.

gfwilliams, 2022-05-03:
> The more authors the more it is likely used.

Nice idea, but I'm not sure it's a good metric. There are quite a lot of apps (especially watch faces) that only have one maintainer and will still be very popular.

Even commit history may not be a good choice - something like a clock may just be really simple and work first time. It doesn't have to include the kitchen sink to be popular

jdekarske, 2022-05-03:
> Sure we can use our own app loader to avoid being in the statistics.
I think any sort of tracking should be strictly opt-in so people don't need to do so. Octoprint and Home Assistant both are, I neglected to pull that info out for my suggestion.

gfwilliams, 2022-05-04:
I do effectively track downloads already since the webserver stores an activity log, but I'm not sure that's a good metric either since:

a) it depends on how many updates an app has
b) an app can get lots of downloads just by having a good description - we're not counting uninstalls

What we could do, as an opt-in, no login solution, is every time a user connects with the app loader, we upload the serial number or MAC address of the Bangle, plus a list of the apps installed - and maybe we drop any data that's more than 6 months old. That's not really personally identifiable, and it'd be kept securely on the server anyway.

We could do likes/comments that way too. To me, that seems preferable as there's no email address/password/other info, and there's no login required. It'd be trivial for me to implement.

nxdefiant, 2022-05-07:
Just for fun counted the authors, this is the top 50:
> 40 locale
13 setting
13 messages
11 boot
10 gpsinfo
10 gbridge
10 dtlaunch
9 widbatpc
9 calendar
9 alarm
8 toucher
8 impwclock
8 gpsrec
8 files
8 compass
8 calculator
7 recorder
7 pipboy
7 numerals
7 miclock
7 mclock
7 marioclock
7 largeclock
7 health
7 cliock
7 boldclk
7 blobclk
7 aclock
6 wohrm
6 weather
6 wclock
6 verticalface
6 vectorclock
6 swatch
6 slidingtext
6 slevel
6 sleepphasealarm
6 scolor
6 sclock
6 pomodo
6 mylocation
6 morse
6 metronome
6 launch
6 ios
6 devstopwatch
6 counter
6 blackjack
6 beebclock
6 antonclk

[num_authors.sh.txt](https://github.com/espruino/BangleApps/files/8644635/num_authors.sh.txt)

thyttan, 2022-11-06:
Now when [sorting by popular apps](https://forum.espruino.com/conversations/380773/) is implemented should this be considered done?

jdekarske, 2022-11-06:
@gfwilliams may want to add a privacy policy depending on your local laws (and/or USA law since the app loader is hosted on Github?). 

https://github.com/espruino/BangleApps/issues/1542#issuecomment-1102759037

Btw, thank you Gordon for adding this! :)

gfwilliams, 2022-11-07:
Ok, thanks - I'll add a link to (and update) our existing privacy policy

closed 2022-11-07 by gfwilliams