espruino/BangleApps #1550: [messages app] Possible memory leak when messages are ignored

jeyemhex, 2022-03-08:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

### Describe the bug
Memory usages grows until the performance of the watch is affected when a notification gets pushed to the watch from Gadgetbridge and is ignored (i.e. message times out and returns to the clock). While the notification icon is present and flashing, memory usage gradually increases over ~ 5-10 minutes until the watch performance is drastically reduced, forcing the user to reset the watch. This can be stopped if the message is read before the memory consumption grows too high, and the memory usage returns to normal levels.

### Expected behaviour
Leaving a message unread for long periods of time should not affect the performance of the watch.

### Other software information
(Not sure how much is relevant)
Apps installed on the watch:
 - Messages v0.24
 - Bootloader v0.42
 - Launcher v0.12
 - Settings v0.42
 - Lock widget v0.06
 - Android integration v0.07
 - Charging status v0.01
 - Bars widget v0.02
 - Minimal analogue clock v0.03

Only tested on Android, gadgetbridge v0.65 is installed from the F-droid store.

### Steps to reproduce
1. Install the relevant apps from the list above onto the watch (No other apps were installed)
2. Send test notification from the gadgetbridge debug menu
3. Ignore the notification until the watch returns to the clock
4. Monitor watch memory usage using the Bars widget over the next 10 minutes, then try to use the watch. 
5. To fix:
    - Reading the message clears the memory and resolves the issue
    - Resetting the watch using the physical button clears the memory, but it will increase again while the notification widget is showing

gfwilliams, 2022-03-09:
Thanks for such a thorough report! This is super useful, I'll take a look today and see if I can reproduce and fix.

alessandrococco, 2022-03-09:
This leak could explain some of the issues reported [here](http://forum.espruino.com/conversations/372748/). I [suspected](http://forum.espruino.com/comments/16403330/) a leak in the app but was unable to test. Great report!

gfwilliams, 2022-03-09:
Just fixed. It seems this was only really a problem for clock apps/widgets that call `Bangle.drawWidgets` repeatedly (that clock calls it every second!). Something like Anton Clock won't do it, *but* you **would** get a small memory leak from every screen lock/unlock as that forces a `Bangle.drawWidgets` so over time it would add up

commit e0937a94 (2022-03-09): messages 0.25: Fix widget memory usage issues if message received and watch repeatedly calls Bangle.drawWidgets (fix #1550)
closed 2022-03-09 by gfwilliams