espruino/BangleApps #1553: Wrong message for required reboot

GrandVizierOlaf, 2022-03-09:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

When uploading a new language (en_US) to my Bangle.js 2 I was asked to reboot with the message `Hold BTN3 to reload`.

This message should only apply to the Bangle.js 1. I have not seen this message before (have always seen `Hold button to reload`) and trying to duplicate the issue was unsuccessful.

I hate to open an issue without reproducible steps, but hope that someone with more familiarity with the code will be able to see how this might occur.

I do bounce around between the main app loader, the development one, and the one from my own github account a fair amount, in case that matters. This was yesterday, but I'm 90% confident I had just loaded an app from dev and then switched to the main loader to update my language, but there's a chance I was loading the language from the dev loader as well.

myxor, 2022-03-10:
This happens for all custom apps (apps which have a web interface on the app loader).

gfwilliams, 2022-03-10:
Thanks - just taken a look and it seems it doesn't always happen for Custom apps. It looks like maybe it's if you upload a custom app *and* you're not already connected to the Bangle (so the app loader can't figure out which one you're connected to).

Will get a fix in soon