espruino/BangleApps #1566: [Question] Howto install SleepPhaseAlarm?

EnerGeh, 2022-03-13:
I'm unable to install the SleepPhaseAlarm App on my Bangle.JS 2. Is there a hidden feature to do so? I have only the like button (heart) but no Upload Button.
I appreciate any help 🤗
[image]

gfwilliams, 2022-03-14:
That's an odd one. If you run the App Loader on another computer, do you have the same issue? I just tried here and the upload button appears for me

EnerGeh, 2022-03-14:
Fair point!
But I just discovered, that the content inside that surrounding frame is horizontally scrollable 🤦
All other apps I installed are with a upload button that doesn't need scrolling. 😩

closed 2022-03-14 by EnerGeh

nxdefiant, 2022-03-14:
Reproducible when Browser is set to mobile layout with width less then ~500px

gfwilliams, 2022-03-15:
Thanks for tracking this down @nxdefiant!