espruino/BangleApps #1579: Run App Replace File Prompt takes a really long time to respond.

Ronin0000, 2022-03-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

When the run app asks you to Replace the file when you press “new” it takes a really long time for the screen to change back to the run app, it also takes a bit of time when you press the other options “yes” or “no”.

gfwilliams, 2022-03-16:
Roughly how long is 'really long' and 'a bit of time'? Are we talking about a second, or more than that?

Ronin0000, 2022-03-17:
I realized that the other options are fine, but the "new" button takes about 4 seconds to get back to the run app screen.

gfwilliams, 2022-03-17:
Have you compacted the storage recently on your Bangle?

I just tried it and it takes 1 second at most here... If your storage hasn't been compacted recently you'll probably be getting all kinds of slowdowns

GrandVizierOlaf, 2022-03-17:
I've been running into this same problem for the last week or two and compacting the storage solved it for me just now.

I don't remember reading about compacting the storage before; is there a guide for when we should be doing it? Is it something that could be tracked on the watch and a tool (a widget, the app loader, etc) could warn when a threshold is met that a storage compaction would be beneficial and run one for the user?

Ronin0000, 2022-03-17:
That seemed to fix the problem. Maybe like @GrandVizierOlaf said we could make an app that compacted the storage weekly at a set time during the night..

closed 2022-03-17 by Ronin0000

gfwilliams, 2022-03-18:
@GrandVizierOlaf I just added something to the Wiki, but it's been discussed a bit on the forum.

But yes, I've actually put some stuff in place recently to allow this to be done (2v13 will have the ability to query storage to see how many deleted files are in it). However it seems there's another (probably more useful) tweak that could be made that'll stop  performance from degrading: https://github.com/espruino/Espruino/issues/2152

So hopefully once that goes in, everything should get a whole lot faster anyway