espruino/BangleApps #1593: Backup: Can not restore files with size=0

nxdefiant, 2022-03-20:
I had to restore the backup for my BangleJS1 today after a firmware update from v10 to v12 somehow made the clock unusable: The clock was still there, but I could not access the launcher any more.

So I tried to restore the backup, but at ~30% it complained about a file with size zero. In my case it was a files named hourstrike.json. I removed it from the zip file and restore worked.

gfwilliams, 2022-03-21:
Do you have the error message that you saw?

To be honest it seems like a bit of a bug that anything is creating a 0 length file.

nxdefiant, 2022-03-21:
I have not copied the exact error message. I just tried to create that scenario, but unfortunately I can not create a new Backup with the BangleJS1, it stops at ~95% and the watch writes "Memory" in red.

commit a09df566 (2022-03-30): ignore backup files of length 0, fix #1593
closed 2022-03-30 by gfwilliams