espruino/BangleApps #1600: [Bordle] Letters not always colored correctly

banglejsdev, 2022-03-21:
### Affected hardware version

Bangle 1, Bangle 2

### Your firmware version

2v12

### The bug

The letter color hints aren't always correct when the answer has duplicate letters. Please refer to the attached screenshot. Note that in row 3, the 'L' in the middle is yellow, but in fact there's only one 'L' in the correct answer ('WOOLY').

Also, sometimes in the final letter selection screen, the border color around a letter is incorrect. I haven't figured out how to reproduce it, but sometimes a letter present in the right location has a dark blue border instead of a green one. (I've seen this happen with 'E').

Regardless, Bordle is great--thanks so much for making this! BTW, I know it'd be a pain but drawing individual characters on the letter grouping screen with appropriate green, yellow, and blue background colors might improve usability even more.

[image]
.

Ronin0000, 2022-03-21:
Also, just incase you do not already know you can call g.dump() to get a image of the screen in the web ide so you do not have to take an actual picture.

banglejsdev, 2022-04-02:
> Also, just incase you do not already know you can call g.dump() to get a image of the screen in the web ide so you do not have to take an actual picture.

Thanks for the useful tip!

banglejsdev, 2022-04-02:
I took a look at the code and see what causes the bug. It turns out it's not due to the answer having duplicate letters but rather that when choosing the color the code only checks to see whether the answer contains a letter:

      else if (sol.includes(c)) col = "#ff0";

Thus it happens whenever the guess has duplicate letters.

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
I think this maybe is solved with this commit: https://github.com/espruino/BangleApps/commit/5152e68c8038087a594ddd13bdf6def171743d1b, not 100% sure.

banglejsdev, 2022-12-02:
Unfortunately, this bug is still open (just checked by running the latest code in the emulator). For anyone else interested in verifying, the target word can be overridden by replacing the following line (87):

this.word = this.words.slice(i, i+5).toUpperCase();

with

this.word = "WOOLY";

bobrippling, 2023-11-13:
The bug is likely in a similar area to #1601 too