espruino/BangleApps #1601: [Bordle] Letter selection border color incorrect

banglejsdev, 2022-03-21:
I just came across a case of the other bug I mentioned in my prior report, namely, that the border color around a letter on the letter selection screen is sometimes incorrect. Note in the second row of the following screenshot that 'T' is in the correct location:

[image]

However, in the letter selection screen, the border color for the letter 'T' is yellow instead of green:

[image]

Incidentally, the correct answer was 'TOADY'. Also, as I mentioned in my other bug report, I've also seen a case where the letter 'E' should have had a green border but had a blue one instead.

Anyway, thanks again for making this great app!

thyttan, 2022-12-02:
Candidate for being closed - leave a comment if you object!

Rationale:
I think this maybe is solved with this commit: https://github.com/espruino/BangleApps/commit/5152e68c8038087a594ddd13bdf6def171743d1b, but not 100% sure.

banglejsdev, 2022-12-02:
Unfortunately, this bug is still open (I just checked by running the latest code in the emulator). The other commit probably fixes issue #1657. I will verify that soon.

gfwilliams, 2025-12-18:
Closing - this seems minor and it seems nobody has been bothered to fix it