espruino/BangleApps #1615: [Close Button] Does not show on 1024 Game.

Ronin0000, 2022-03-24:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

The Close Button Should show on every screen that is not the clock. I am not sure why it does not show on this app only.

gfwilliams, 2022-03-24:
The close button? You mean the newish widget that displays the close button? If apps don't display widgets then you won't see it - that could be the issue?

Ronin0000, 2022-03-24:
Yes the close button:
[image]
It should show on every app that shows the widgets except the clock. Here it does not show on this game:
[image]
But on every other app that shows widgets except clocks it does show the button:
[image]

rigrig, 2022-03-24:
This happens because the game says it is a clock:
https://github.com/espruino/BangleApps/blob/0a347ee03166662d5dbed0eb3d7ba292079dd15f/apps/game1024/app.js#L479-L480

chiefdaft, 2022-03-25:
Removing line _Bangle.setUI("clock");_ let's the close button to appear and it can close the app if you don't hit the undo button accidentally. I'll fix this issue.

Ronin0000, 2022-03-25:
I see now, Thanks!

closed 2022-03-25 by Ronin0000